Ultimaker/CuraEngine #1073: [ERROR] Trying to retrieve setting with no value given: 'mesh_rotation_matrix' when using command line

WolfgangFahl, 2019-06-12:
**Application Version**
4.1.0

**Platform**
MacOS High Sierra 10.13.6

**Printer**
Geeetech A10

**Action**
```
curadir="/Applications/Ultimaker Cura.app/Contents"
curae="${curadir}/MacOS/CuraEngine"
curar="${curadir}/Resources/resources"
printer="$curar/definitions/Geeetech_A10.def.json"
"$curae" slice -l /tmp/$design.stl \
  -j "$printer" \
  -s extruder_nr=0 \
  -s machine_nozzle_size=0.4 \
  -s material_diameter=1.75 \
  -o /tmp/$design.gcode
```

**Actual Results**
[ERROR] Trying to retrieve setting with no value given: 'mesh_rotation_matrix'

**Expected results**
slicing, no error message

rburema, 2019-06-12:
Hi @WolfgangFahl,
Your definitions folder should contain the fdmprinter.def.json, which should have `mesh_rotation_matrix`.

nez, 2019-11-30:
I set `-s mesh_rotation_matrix="[[1,0,0], [0,1,0], [0,0,1]]"` (taken from cura.log) and still I get the same message.

Also, shouldnt the identity matrix be the default value for mesh_rotation_matrix?

WolfgangFahl, 2019-12-01:
I think this is too complicated and a showstopper for trying things out. I didn't bother to try again.

rburema, 2019-12-03:
Hi @nez 

> ... and still I get the same message.

Did you specify the `-l` option _after_, or _before_ the added `-s` settings?

> Also, shouldnt the identity matrix be the default value for mesh_rotation_matrix?

I suppose that would be useful. The command line usually doesn't get much attention though, since the 'normal' way to call it (at the moment at least, this has been the case for a long while) is through some sort of front-end, via libArcus.

As far as I understand, the command-line was originally meant for debugging (but we since have other means, for instance, the 'start the engine manually' command line option in Cura itself).

tdeck, 2024-09-29:
i spent some time this weekend really trying to get CLI slicing to work and ran into the same issues as OP. But I had a breakthrough.

This error **doesn't mean** there's no value specified for the setting (contrary to its wording). It means there's no **definition** for the setting in a JSON file. You need to provide a def.json file for the printer and for the extruder. That's true even if you're specifying every single setting on the command line because you copy-pasted them out of the Cura logs.

I've written some quick and dirty shell scripts [in this repo](https://github.com/tdeck/cli-slice) to make this work which you can use as inspiration (you'll likely need to change paths). Running `last-cura-settings` will output the last set of slice settings used to slice a model in the GUI. You can save these in a .set file and then use them in `slice-with-cura` to slice models.

One thing I discovered is that Cura seems to pre-transform the STL mesh in the GUI before piping it to CuraEngine. This means that `mesh_position_...` and `mesh_rotation_matrix_...` will always be the default values. Your meshes will not be centered when slicing on the CLI by default because Cura's centering happens in the GUI and doesn't get set as part of the settings, so slicing the exact same model with this CLI method and the GUI method will result in a different position on the build plate (or, as I often find, off the build plate!).

Hopefully this is helpful to other folks. I'm not sure how much further I'll take my own experiments but at least I am able to get past the errors.