Ultimaker/CuraEngine #1075: [Build] MSVC build broken

liuyazui, 2019-06-14:
**Application Version**

CuraEngine4.0.0, libArcus4.1.0, protobuf3.8.0, vs2019x64, CMake3.14.3

**Platform**

Windows10 visual studio 2019 x64

**Additional Information**

The dependent libraries (protobuf and libArcus) have been compiled successfully. However, when I build CuraEngine, 'clipper' can be compiled successfully, and I get the 'clipper.lib'. When I build '_CuraEngine', I got some issues below:
1>D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(185,5): error MSB4018: The "GetOutOfDateItems" task failed unexpectedly.
1>D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(185,5): error MSB4018: System.NotSupportedException: The given path's format is not supported.
1>D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(185,5): error MSB4018:    at System.Security.Permissions.FileIOPermission.EmulateFileIOPermissionChecks(String fullPath)
1>D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(185,5): error MSB4018:    at System.Security.Permissions.FileIOPermission.QuickDemand(FileIOPermissionAccess access, String fullPath, Boolean checkForDuplicates, Boolean needFullPath)
1>D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(185,5): error MSB4018:    at Microsoft.Build.CPPTasks.MsbuildTaskUtilities.GetWildcardExpandedFileListFromMetadata(IBuildEngine buildEngine, ITaskItem item, String metadataName, Boolean convertToUpperCase)
1>D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(185,5): error MSB4018:    at Microsoft.Build.CPPTasks.GetOutOfDateItems.Execute()
1>D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(185,5): error MSB4018:    at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute()
1>D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(185,5): error MSB4018:    at Microsoft.Build.BackEnd.TaskBuilder.<ExecuteInstantiatedTask>d__26.MoveNext()
1>Done building project "_CuraEngine.vcxproj" -- FAILED.

Ghostkeeper, 2019-06-14:
The error messages you're getting are not in our code. They don't tell me much. I see permission errors (maybe the files are restricted to be modified by admins only?) and I see that GetOutOfDateItems failed for some reason which is how MSVC detects which files it has to compile and which files haven't changed since the last compilation.

I'm guessing that something is wrong in your MSVC project. We generally don't maintain MSVC compatibility for CuraEngine, but it should still work with nmake as far as I know. Could you try with nmake generators instead of a vcxproj file?

rburema, 2019-06-20:
We've recently udpated our documentation for Windows-builds as well. See [here](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Windows#curaengine).