Ultimaker/CuraEngine #1081: Possibility to add polar coordinate space?

BKLronin, 2019-06-20:
As the title says, can someone tell me if it is technical possible to replace the cartesian space with a polar space? 

Thank you in advance.

smartavionics, 2019-06-20:
Hello @BKLronin . Not sure what you mean there. Are you referring to the input space (i.e. replace models 
specified as x/y/z with mag/deg/?) or the output space (some form of polar coordinate version of gcode)?

BKLronin, 2019-06-20:
Hello @smartavionics,

I mean placing normal models that have been created in xyz on  a polar space so that the resulting gcode would be natively  in polar coordinates without post processing? So that the engine directly outputs into polar. Postprocessing needs a lot of segmentation and a lot of small moves increasing the file size to GB on larger models.

smartavionics, 2019-06-20:
OK, so do you have a syntax for polar gcode?

BKLronin, 2019-06-20:
I got the general transformation from my debug python program. That would be without segmentation. 

```
def polar(x,y):
        #radius calculate
        radius = math.sqrt(x ** 2 + y ** 2)
        #angle calculate
        angle = math.atan2(y, x) * 180.0 / math.pi        

        return radius, angle
```

smartavionics, 2019-06-20:
Sure, but the slicer's output is going to be the input to a printer and that printer (presumably) has a language it understands. Do you have any info on that or is this all just speculation?

BKLronin, 2019-06-20:
The printer runs the klipper firmware so a standard X,Y,Z Gcode in marlin flavor is sufficient. Iḿ trying to implement this with Kevin O Connor from the KLipper project currently but Iḿ more the mechanical engineer than a programmer so my programming skills are limited and Kevins time is as well. I got a full working polar printer for testing and the ability to debug the firmware itself with incoming gcode.

smartavionics, 2019-06-20:
Ah, OK. Well, I can say with 99.9999% certainty that Ultimaker won't be interested in working on this but I may well look into it when there is a printer language  definition or at least a good description of how the printer expects the input to look like.

BKLronin, 2019-06-20:
Thank you very much for your assistance. It seems to work now on the firmware level :)

closed 2019-06-20 by BKLronin