Ultimaker/CuraEngine #1083: [4.1]

rgov, 2019-06-22:
I'm trying to use CuraEngine to generate an efficient path that fills in a 2D polygon, like a pen plotter shading in a shape, or a Roomba trying to vacuum an entire room.

To start with I am attempting to slice a 3D file with 100% infill and no supports or plate adhesion. I'm testing with the [Utah teapot](https://commons.wikimedia.org/wiki/File:Utah_teapot_%28solid%29.stl).

When I set the settings as desired in Cura and enter Preview mode, I can take slices and it does look like it is solid:

[image]

Here is my JSON file configuring these settings:

```json
{
  "version": 2,
  "name": "Generic Printer",
  "inherits": "fdmprinter",
  "overrides": {
    "adhesion_type": {
      "default_value": "none"
    },
    "machine_start_gcode": {
      "default_value": ""
    },
    "machine_end_gcode": {
      "default_value": ""
    },
    "infill_pattern": {
      "default_value": "lines"
    },
    "infill_sparse_density": {
      "default_value": 100
    },
    "support_enable": {
      "default_value": false
    }
  }
}
```

I run CuraEngine with `CuraEngine slice -j "printer.def.json" -l ~/Downloads/Utah_teapot_\(solid\).stl > /tmp/out.gcode` and then view the GCode file with [this online viewer](http://gcode.ws).

At the widest layer of the teapot, the GCode viewer suggests that the printer would only make 5 stripes to fill it in:

[image]

Am I failing to set this up correctly?

Here is my [Gcode file](https://github.com/Ultimaker/CuraEngine/files/3317266/out.gcode.txt).

rgov, 2019-06-22:
Other questions:

* What setting dictates the spacing of the infill lines? I would guess it's related to the thickness of the filament. I.e., in the pen plotter scenario, I want the fill lines separated by the width of the pen; in the Roomba scenario, it would be the diameter of a Roomba. 

* Can I disable the inner/outer walls and only generate a fill that goes right up to the boundary of the shape?

* How do I use the SVG debug functionality to view the paths without using a GCode visualizer?

rgov, 2019-06-22:
When I export GCode from Cura itself, the resulting file is 1.8 MB, and the online viewer shows a much denser fill. But also, if I reduce my infill percentage to 50%, the file gets smaller, so my infill setting *is* having an effect.

rgov, 2019-06-23:
A solution seems to be:

```json
    "bottom_layers": { "default_value": 999999 },
    "top_layers": { "default_value": 0 },
```

These are conditionally set in `fdmprinter.def.json` when `infill_sparse_density == 100`, but this doesn't seem to get evaluated.

rgov, 2019-06-23:
I'm satisfied with how this turned out now.

closed 2019-06-23 by rgov

Ghostkeeper, 2019-06-26:
> Am I failing to set this up correctly?

CuraEngine actually only reads the leaf-settings, not the parent settings such as `infill_sparse_density`. CuraEngine only reads `infill_line_distance`. Cura's front-end sets the line distance such that the desired density is achieved, given the infill line width and pattern.

> What setting dictates the spacing of the infill lines?

`infill_line_distance`.

> Can I disable the inner/outer walls and only generate a fill that goes right up to the boundary of the shape?

Yes, set `wall_line_count` to 0.

> How do I use the SVG debug functionality to view the paths without using a GCode visualizer?

The SVG debug functionality is meant for programmers who are able to compile CuraEngine. If you can compile CuraEngine, you can output the `Polygons` instance to a file by calling:

```
SVG svgfile("file.svg", aabb);
svgfile.writePolygons(polys);
```

> When I export GCode from Cura itself, the resulting file is 1.8 MB, and the online viewer shows a much denser fill. But also, if I reduce my infill percentage to 50%, the file gets smaller, so my infill setting is having an effect.

Yeah, that's because the front-end will adjust the Infill Line Distance setting automatically for you.

> A solution seems to be:

Typically, setting the skin thickness to be infinite results in a better print quality than actually increasing the infill density.

> These are conditionally set in `fdmprinter.def.json` when `infill_sparse_density == 100`, but this doesn't seem to get evaluated.

Only the front-end can evaluate these expressions. They are Python expressions after all.