Ultimaker/CuraEngine #1084: Feature request: combing only over lines/material

richfelker, 2019-06-23:
Currently, retraction combing treats the entire abstract interior of the solid model as a candidate for combing moves. In particular, it can move across the interior of the model even if there is 0% infill, and across unprinted space between infill lines.

This is fine for materials that don't tend to to ooze/string, but for materials that do, it results in stringing/ooze inside the model. This:

- causes unpredictable loss of material, resulting in underextrusion after the move.
- for flexible-material prints, the strings alter the "texture" of the object felt when squeezing it

The stringing issue is less pronounced with high infill percent, since the oozed material gets wiped each time it hits an infill line, but either way the unpredictable loss of material happens. With flexible materials, this is a serious problem. You can'd just do "extra prime after move", because if material wasn't lost during the move, this will create a serious over-pressure issue in the bowden resulting in oozing *next time* a travel is attempted, even with retraction.

In addition, the main motivation for using the "noskin" or "infill only" combing modes is to avoid ugly scarring across the skin layers from traversing them along a different path than the original lines.

All of this would be mitigated by a combing mode that forces combing moves to follow the path of existing lines, rather than performing arbitrary moves across the interior of the "solid" model slice. The presence of an existing line under the nozzle prevents oozing almost entirely - sufficiently that it doesn't matter, anyway.

richfelker, 2019-06-23:
Note that with 0% infill or with infill patterns that are not connected (e.g. crosses, I think), this would prevent all/most combing travel across infill region and would force combing to take place around the perimeter, or force retraction to be used instead when that's not possible.

smartavionics, 2019-06-24:
Hello @richfelker, I understand what you are saying there and it sounds like a nice idea. Very hard to implement though because the combing engine would have to know the positions of the infill lines on the previous layer (which it currently doesn't and making it happen would require the layers to be processed sequentially which it no longer does since becoming multi-threaded).

What happens if you disable combing completely and just use retracted travel moves?

richfelker, 2019-06-24:
It only needs to know the positions of lines in the *current* layer, not the previous one. If you comb over existing lines with layer-height spacing between the nozzle and them, using a material that tends to ooze, you'll draw solid lines until melted material is depleted - the tendancy of the material to bond to itself *pulls it out*. Combing only works right when the gap is 0.

Disabling combing does "work", but depending on the model can create thousands of retractions. Limiting the number of retractions is not an option because any omission of retraction will ooze material. Also, with a bowden extruder, very long retractions (I use 15mm) are necessary to compensate for compression of the material, making it really slow (on top of the already slow print speeds needed for these materials).

So, combing is highly desirable to leave enabled, and even now with the problem described in this issue, the tradeoffs for disabling it are usually worse.

smartavionics, 2019-06-24:
> It only needs to know the positions of lines in the current layer, not the previous one. If you comb over existing lines with layer-height spacing between the nozzle and them, using a material that [...]

Of course, I was being stupid. So for the idea to work, the infill needs to be printed before the walls. It's still a lot of work to implement and there would almost certainly be quite a cost in terms of slicing time but it could be done. I will think about it!

richfelker, 2019-06-24:
Thanks.

Yes, to comb over infill to reach the walls, the infill would have to be printed before the walls. However you actually probably want the inner walls already there to comb over between one line of infill (or skin) and the next - this is actually an area where I saw heavy retraction with combing turned off. For skin maybe it would be fixed by setting the max travel with no retraction to the nozzle/line width rather than 0, but for infill lines I think it would always happen.

For a simply connected region, if you've already drawn a full perimeter, combing over infill is actually just an optimization vs going around the perimeter. I wonder if an easy short-term approach to this would be just adding a "comb over non-skin walls only" mode. This would require retractions to get between the walls of "holes" and the outer perimeter of a part, but that's probably not so bad.

richfelker, 2019-06-24:
To clarify, "comb over non-skin walls only" would be "noskin" but with the added constraint "noinfill" as well. A choice to allow combing over everything but infill (i.e. even skin and outer walls) might also be useful too.

Ghostkeeper, 2019-07-03:
We won't have time to implement this any time soon, sorry. We'll keep it open for when we get to it later or someone finds a reason to increase the priority, but it's not going to go on our priority list now.

richfelker, 2020-01-21:
In the mean time, a workaround I've found that seems tolerable without disabling combing entirely is setting a very low `retraction_combing_max_distance`. And I think this may also work around a bug in `fill_outline_gaps` where retractions are skipped between tiny (smaller-than-line-width) components - IIRC the math for computing the region in which combing is allowed for these is erroneous and allows "combing" *outside of the solid*.