Ultimaker/CuraEngine #1085: Will the infill path ever exit the polygon?

rgov, 2019-06-24:
I used CuraEngine to generate the infill on a polygon shaped like a skull and crossbones, and then plotted the traversal path of the extruder from the generated GCode:

[image]

I notice that the extruder's path never exits the bounds of the polygon, even when not extruding filament. Is this always true?

smartavionics, 2019-06-25:
Depends on whether combing is enabled. If combing is not enabled, travel moves will be direct so they may cross the boundaries of the polygon.

rgov, 2019-06-25:
That's awesome, thanks! Here's with combing disabled:

[image]

closed 2019-06-25 by rgov