Ultimaker/CuraEngine #1086: Feature request: fill_perimeter_gaps with thin lines

richfelker, 2019-06-26:
This issue is most prominent when printing walls thinner than `wall_line_count*wall_line_width` whose thickness is not a multiple of `wall_line_width`: the fill added by `fill_perimeter_gaps` will be a very slow-printing zigzag. Instead, it could be printed as a thin line (thinner than nozzle width) simply by lowering the flow; the existing wall lines on both sides should ensure that the material fills the space correctly, except in the special case where the entire wall thickness is less than 2 (or even 1) wall line.

Also related: it may be desirable to be able to control `fill_perimeter_gaps` separately for top/bottom layers vs internal ones, but I'm not sure if that's a distinction that's easy to make. For many prints, printing 1mm walls with two 0.4mm lines and nothing in between would be fine (and save lots of print time) except that the top and bottom layers absolutely need an extra (thin) line or zigzag in between so that there's no hole into the interior of the object.

smartavionics, 2019-06-27:
Hello @richfelker , my Cura builds implement the wall gap filling and thin walls as you are suggesting. They can be installed alongside Ultimaker's releases. You can find builds for Linux and Windows at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

It's a work in progress but for a lot of prints works OK. Here's an example that shows variable width lines..

[image]

Ghostkeeper, 2019-06-27:
This is already implemented with the Print Thin Walls feature. Unless you meant this to be a bug report. I think that's what you meant, so I'm closing this as a duplicate of https://github.com/Ultimaker/Cura/issues/5110.

We've made improvements with the bug for Cura 4.2. It produces a similar result as in SmartAvionics' build but does hold a minimum thickness in order to keep flowing material properly. This is how it looks:
[image]

It still doesn't work with all cases we could find, so there's more work to be done, but please continue discussion over at https://github.com/Ultimaker/Cura/issues/5110.

closed 2019-06-27 by Ghostkeeper

richfelker, 2019-06-27:
OK. It doesn't work for me so I'll continue on the other thread. I think this is possibly also a documentation issue since I understood `fill_outline_gaps` to be about thin walls (parts of the sliced region narrower than a single line), not using thin lines internally on thick walls. I've also had trouble with this option that led to me turning it off, which I'll try to describe there or in a new report.