Ultimaker/CuraEngine #1088: Feature request: cool_min_layer_time only below certain Z position

richfelker, 2019-07-01:
In my experience, certain prints need `cool_min_layer_time` to prevent severe curling at corners, but only below a certain Z position, typically around 3-4 mm. This is presumably because cooling effeciveness is reduced in a vicinity of the heated bed. Rather than slow down the entire print with an increased layer time, it would be nice to have an option to apply a different mimimum below a certain height/layer-number, or to smoothly decrease the minimum with increased height.

richfelker, 2019-07-01:
Note that `speed_slowdown_layers` can be used as a substitute for this in some instances, but it also severely slows down the fill speed for skin, resulting in an overall significant increase in print time if you have large skin surfaces near the bottom of the print or a large number of bottom layers.

An alternative workaround is probably a 4mm high junk object next to the real object to print, but that's something of a waste of material and may produce stringing depending on material and other settings.

Ghostkeeper, 2019-07-02:
You're using the wrong tool here. Instead of Minimum Layer Time, which is meant to allow the previous layer to cool down before putting the next one on, you're looking for Initial Layer Speed, which is meant to print slower so that the current layer stays warmer for longer which anneals the material. The effectiveness of Number of Slower Layers, as you describe, is dependent on the annealing. You specifically *don't* want stuff to cool down.

In my experience the minimum layer time doesn't apply to the first few layers because:
* Those layers are almost always the largest layers in the entire print, because other layers need to rest on them (with support and all).
* They tend to print the slowest because the printer is very careful for bed adhesion.
* They tend to have lots of skin, so it's also filled the densest which takes longer to print.

It's extremely rare to have a first layer on the order of magnitude of 10s, let alone to have higher layers take longer than that as well then. That's why we'll reject this feature request.

richfelker, 2019-07-02:
This request is not about the first layer and is not about slow cooling to anneal. It's about the fourth (or whatever the first after bottom skin finishes) through approximately 20th layers. I can provide examples if needed. I think you've closed this based on a complete misunderstanding of what the request was.

richfelker, 2019-07-02:
Simple testcases I can provide are hollow 18mm icosahedron and dodecahedron sitting on one face. All layers but top/bottom print in less than 7 seconds, most in about 3-4 seconds. Near the build plate, either because of heat from the build plate or some other physical effect I don't understand (maybe air flow properties from confined space below nozzle?), the corners curl badly and produce irregular edges. After getting up about 4mm, the problem goes away, despite the layer time not having increased significantly. Increasing min layer time to 6 or 7 makes the problem go away, but makes the top 80% of the print horribly slow for no reason.

For these particular cases, increasing `speed_slowdown_layers` could achieve the desired effect, but it's a hack and it wastes time on the second, third, etc. skin layers that don't need to be printed slowly.

Ghostkeeper, 2019-07-03:
Could it be that your walls are so thin that the sharp contrast between strong 100% filled bottom skin and a single empty contour causes the warping? That normally happens at any height.

Ghostkeeper, 2019-07-03:
I still think that your use case is very specific to your model. The minimum layer time effectively slows down the nozzle, that's all it does. Slowing down the nozzle beneath a certain layer already exists through Initial Layer Speed.

If you really want certain layers to be printed more slowly specific to your model, use a cutting mesh. That's what the cutting meshes are meant for.

richfelker, 2020-01-20:
I finally got around to experimenting with rafts, which I'd written off as a waste of material for folks with poor-adhesion bed materials, and I think they might actually be the right solutions for problems like mine - maintaining strong adhesion while shielding the first few tens of layers from warping due to the heat of the bed.

Ghostkeeper, 2020-01-24:
Agreed, looking at this again, we're probably not going to implement this since it has no relation to the intended use of `cool_min_layer_time` (as per my first comment). Sorry.

closed 2020-01-24 by Ghostkeeper