Ultimaker/CuraEngine #1092: Feature request: brim gap

richfelker, 2019-07-09:
I usually don't use brim, but found myself wanting it for a few prints with lots of small pieces attached to the bed only by small surfaces. However brim is damaging to remove, and I want surface quality on these parts. My thought was to setup a brim gap comparable to a support gap, maybe slightly smaller, to provide some hold but leave it easy to remove without damage to the printed part. However there seems to be no such option.

Using a skirt instead, and reducing the skirt gap to 0.1 mm or so, works for a single convex outline, but doesn't go between the small parts like a brim would.

I haven't yet looked at the relevant brim generation code but I suspect a gap would be easy to add, just a matter of applying offset to the polygon object(s).

Ghostkeeper, 2019-07-10:
Perhaps it'd be more logical to have an option for the skirt to not make an almost-convex shape.

Never really heard about the brim being damaging to remove though. Removing the brim with a knife leaves as nice a surface as a print with skirts.

smartavionics, 2019-07-10:
It just so happens that my Cura builds support a brim gap setting. I use it to space the brim very slightly away from the outside of the model so that it is easy to remove but I still get the thermal benefit of having a brim (reduces the amount the corners warp). As always, you can find my releases at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0.

richfelker, 2019-07-10:
@Ghostkeeper, It's not that they're "hard" to remove, but damaging. My model meets the bed in overhangs like:

```
\____/
```

and with a brim meeting it at zero gap, it will effectively turn into:

```
)_____(
```

This could be mitigated with careful post-processing, but that would damage the surface finish, and isn't much fun when you have lots of tiny parts as I described. I think adding a tiny gap would prevent the brim from distorting the wall at the base while still providing assistance with adhesion.

AbeFM, 2019-07-23:
I have seen bottoms come off when tugging on a brim unwisely.

I thought it was in CURA but I guess it was S3D, I would use a skirt spacing of ~1/2 a line width and they worked excellently, it was my default mode. But the skirts were just spaced out brims like GK is saying. 

TL;DR: +1

Timocop, 2019-07-27:
This is also realy usefull for people that use thick initial layers. I use thick inital layers and its so painfull to remove brims... if the brim even comes off that is. Because that, i dont even use brims anymore unless its ABS because its soft.

Me once i removed the thick brim from my model:
[image]