Ultimaker/CuraEngine #1121: Compilation error on Ubuntu 19.10 (devel)

thopiekar, 2019-08-23:
Hey!

here another broken build. It only happens on the most recent Ubuntu distribution, which is still under development. The compiler used is GCC 9.2.x.

https://launchpadlibrarian.net/438699413/buildlog_ubuntu-eoan-amd64.cura-engine_1%3A4.2~201908231903~rev2184~pkg315~ubuntu19.10.1_BUILDING.txt.gz

Build environment: PPA (launchpad.net)
OS: Ubuntu 19.10 (development branch)
Architecture: Any (i386, amd64, armhf, etc...)

LipuFei, 2019-08-23:
Same issue as https://github.com/Ultimaker/CuraEngine/issues/984 Current OpenMP pragmas are not compatible with GCC 9.

closed 2019-08-23 by LipuFei

thopiekar, 2019-08-23:
And also see a PR which should solve the problem. Great!
Any plans to get the PR merged? 😃

Ghostkeeper, 2019-08-25:
Yeah there has been activity in that topic last week too.