Ultimaker/CuraEngine #1126: Weak Prime Tower with Variable Layer Height

AbeFM, 2019-08-30:
**Application Version**
~4.2.1 (Smart Avionic's version)

**Platform**
win10

**Steps to Reproduce**
Slice model with Variable Layer Heights and Prime Tower (especially one where the min volume just brings it to an integer 1 or 2 perimeters)

**Actual Results**
Thin, under-extruded layer-groups in tower leading to weak towers.

**Expected results**
Magic towers that always work no matter what settings I use!

**Additional Information**

[image]

(this stuff has to print hot so I get these terrible strings...)

[image]

There's a few options on how to deal with this and I don't yet have an opinion on it.

AbeFM, 2019-08-31:
Additionally: Higher nozzle pressure leads to worse stringing on smaller layers.

Ghostkeeper, 2019-09-04:
The priming volume is based on the assumption that the layer height is equal to the `layer_height` setting, which is incorrect if using adaptive layers. As a result, it'll not reach the desired priming volume for thinner layers and will prime way more for thicker layers.

AbeFM, 2019-09-04:
So incorrect priming is desired behavior?

It sounds like you're saying I have it backwards, so not only is the tower
weak in some levels but you're also underpiming in places as well?

It would seem to me you'd either:

1) over extrude on thin layers to achieve "minimum prime volume" so the
user gets the volume requested. Your description makes it sound like this
happens - it's the opposite of my impression but I need to check more
carefully before I'd insist. I assume you're right. My feeling was a volume
enough to fill the tower at standard layer height is extruded and the
thicker layers had gaps.

2) remove a wall from the prime tower to prevent massive over extrusion on
layers that don't have room for the minimum volume, allowing heavy priming
for all layers.

 (maybe this is already done in terms of volume, but the diameter should be
adjusted to accommodate. If the outer wall is three perimeters maybe it
should only be two on the thinnest layers.)

I can probably do some testing today to see what's really going on with
more exaggerated changes.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Ghostkeeper, 2019-09-06:
> So incorrect priming is desired behavior?

No, no, you understood me wrong. I'm not saying it's desired behaviour. It's definitely a bug here. I'm saying that the prime tower volume is making the false assumption that layers have the thickness of `layer_height`. As a result, the prime tower will **not** do as desired.

The prime tower will currently create exactly the same amount of perimeters on every layer. This would need to change for variable layer height to achieve the minimum prime volume. That's similar to your option 2.

I believe though that the prime tower WILL extrude precisely enough material to fill the layer height and line width. In theory this should not underextrude or overextrude. You can see this in layer view: the lines would appear wider if it's overextruding or thinner if it's underextruding, but it appears to be correct. In practice, it will not have primed enough so maybe it will be underextruding.

AbeFM, 2019-09-06:
>It's definitely a bug here. I'm saying that the prime tower volume is making the false assumption that layers have the thickness of `layer_height`. As a result, the prime tower will **not** do as desired.

Well, that's about all I was trying to achieve by mentioning it, now it has an issue number.

> 
> The prime tower will currently create exactly the same amount of perimeters on every layer. This would need to change for variable layer height to achieve the minimum prime volume. That's similar to your option 2.

I would suspect that even just going around the tower an extra time and letting the extra unsupported layers print as mid-air strings which  fall would be better than a tower that fails, and easier to implement. Probably just putting out more material on thicker lines would work.

> 
> I believe though that the prime tower WILL extrude precisely enough material to fill the layer height and line width. In theory this should not underextrude or overextrude. 

Testing now.

AbeFM, 2019-09-06:
Slicing doesn't appear to have thicker lines on screen:

[image]
(PVA extruder excluded from preview for clarity)
[image]

(I just hid upper layers, but I see no obvious gap in either thick or thin layer)

Ghostkeeper, 2019-09-09:
> I would suspect that even just going around the tower an extra time and letting the extra unsupported layers print as mid-air strings which fall would be better than a tower that fails, and easier to implement.

Yeah the prime tower since version 3.5 was designed to be basically one shell and for the rest to "purge in the centre". However I figured that when purging in the centre we might as well keep purging along the side anyway. It was easier to implement and has the potential to make the tower stronger. So printing extra perimeters is no problem if they fall down the middle.

AbeFM, 2019-09-10:
I believe these extras would be on the inside - at least, my issues have been with the outer layer.

I did this a while ago and forgot to post - sorry for the delay, at least it looks like no one did anything on account of my being slow. :-)

Close up of a tower with a lot of variance - 0.1mm-0.32mm layer heights
[image]

MOAR PICS:
https://photos.app.goo.gl/4DKpZR7NMMiyJDC38

I think my conclusion is that if you're way under-primed, you need to use extra restart distance to keep your towers from failing. It looks like the variable heights are reasonably accounted for because the walls are not over/under-stuffed. I don't know if the right amount of priming is happening on the thinner layers, but as I'm likely over-priming anyway to keep the tower reliable, it is not an issue.

I feel this is a caution, but a non-issue, and could be closed. If anyone would like more investigation tell me and I'll try to work it in.