Ultimaker/CuraEngine #1127: G-Code Output not working - No Errors Thrown

LetsPrint3D, 2019-08-31:
I've moved CuraEngine.exe and it's DLL dependencies out of the latest Cura folder to run standalone. I placed the latest "fdmprinter.def.json" in the same directory and used the following command line...

curaengine slice -v -p -j "fdmprinter.def.json" -o "test.gcode" -s infill_line_distance=0.8 -l "test.stl"

It appears to be working for a moment, then it just finishes without throwing any errors or generating the .gcode file. If I omit the -j parameter however, it does generate an empty .gcode file.

So as far as I can tell, the problem appears to be with using -j "fdmprinter.def.json". In case it wasn't looking in the CWD, I tried setting the ENV variable CURA_ENGINE_SEARCH_PATH and pointing it to "X:\CuraEngine" but unfortunately that hasn't solved the problem either.

Ghostkeeper, 2019-09-04:
Most likely it couldn't resolve the fdmprinter.def.json file. CuraEngine uses both the stdout and stderr stream because the stdout is used for g-code output when running via command line without -o command. That means though that just piping the stdout will not produce any output if there's an error in generating.

LetsPrint3D, 2019-09-04:
@Ghostkeeper Thanks so much for the information. Does CuraEngine.exe not look in the root directory for "fdmprinter.def.json"? It did suggest using the system PATH variable to set external directories if applicable, but tried this as well to no avail. Bit stumped as to hunting down this problem without any errors to go on.

Ghostkeeper, 2019-09-06:
The `-j` parameter is required if slicing via the CLI. If it's not provided via `-j` and not via `--connect`, then CuraEngine will not know of any settings to slice with. So there is not a hard-coded `fdmprinter.def.json` in the source code that it will look for if the parameter is not provided.

The parameter supplied with `-j` will be relative to your cwd. So if you were to call these commands:

    cd C:\Users\john\projects
    "C:\Program Files\Cura\CuraEngine.exe" slice -j fdmprinter.def.json -o test.gcode -l test.stl

Then the `-j fdmprinter.def.json` will refer to the file `C:\Users\john\projects\fdmprinter.def.json`. Instead you should probably call something more like:

    "C:\Program Files\Cura\CuraEngine.exe" slice -j "C:\Program Files\Cura\resources\definitions\fdmprinter.def.json" -o test.gcode -l test.stl

The `CURA_ENGINE_SEARCH_PATH` variable is used only to find where CuraEngine is, not where the JSON files are.

LetsPrint3D, 2019-09-07:
So I have been at this for a week, pulling my hair out trying to figure out what I am doing wrong. In a last ditch effort, I tried out an older version 4.0 and it worked instantly the first time...

Seems to be some sort of problem with CuraEngine CLI on 4.2.1 because this is the only output I get from it, and nothing happens.

[image]

Using the same exact commands in 4.0 gives plenty of information and works perfectly. Regret spending a week trying to figure this nightmare out, but at least I finally got it.

nallath, 2019-09-09:
We had a fix regarding this a about 1,5 week ago (https://github.com/Ultimaker/CuraEngine/commit/5ecaa924dfb0fc64c3fcc5d3215d8b9c320dd9d6). Could you check if that solves it for you?

Ghostkeeper, 2019-09-09:
That would explain why I couldn't see any problems with it on Linux.

closed 2024-04-05 by wawanbreton