Ultimaker/CuraEngine #1132: Feature request: slowing faster segments first for cool_min_layer_time

richfelker, 2019-09-07:
Given for example an outer wall speed 30 and inner wall/infill speed 60, `cool_min_layer_time` slows them proportionally, e.g. to 20 for outer wall and 40 for inner wall/infill. This results in:

1. Failure to meet `cool_min_layer_time` when `cool_min_speed` is in use, despite having plenty segments, possibly even a majority, still faster than `cool_min_speed`. And
2. Very low outer wall speeds that **don't help cooling** because the slow speed results in heavy heat transfer from the nozzle to existing material.

It would be much better, in my example case, to reduce inner wall/infill speed down to 30 first, and only reduce them all further if `cool_min_layer_time` is not met.

I suspect this might clash with use of `speed_equalize_flow_enabled`, but I've found that feature doesn't work anyway (it produces speeds way too high for material to adhere correctly), and for users of printers with linear advance firmware, it's not necessary.

richfelker, 2019-09-07:
Note that with this change, setting `cool_min_speed` the same as outer wall speed would be completely viable, and would greatly improve outer surface consistency.

Ghostkeeper, 2019-09-09:
I don't see why this would clash with Equalize Filament Flow. Depends a bit on how we'd implement it I guess.

Ghostkeeper, 2019-09-10:
After discussing this at work, we've decided that we're not going to give any priority on this. The reason was something I hadn't thought about before: If the minimum layer time is being reached, chances are that you'll only have walls left and there will not be a lot of speed variation in that layer anyway.

It's still a valid idea though. I hope we get to it at some point. The entire minimum layer time feature is a bit wonky.

richfelker, 2019-09-10:
Thanks for the reply. For the time being I've just increased my outer wall speed to match inner/infill, and it doesn't seem to hurt print quality, so I got faster printing as a bonus. But this would still be nice to have.