Ultimaker/CuraEngine #1137: When Infill thickness <> Layer Height

radekkpl, 2019-09-18:
**Application Version**
4.2.1

**Platform**
Windows

**Steps to Reproduce**
Print any model with settings:
 - Layer Height 0,05mm
 - Infil Layer Height 0,1 mm (or better 0,15mm)
 - You can add decent infill overlap, but even 80% overlap will look like at the image

**Actual Results**
Extruded amount is correct (3x more then on single layer), but the slicer thinks, that material will be increased horizontally (line width) in single layer, not through 3 separate layers and keeping line width as expected.

This will result in no connected infil to the wall, because the centerline of the infill is calculated on wrong infill line width. (see cura slicing layer image)

**Expected results**
The offset from the wall will be kept, so infill will conect to the wall.
This is printed with 80% infill overlap.
[image]

First single layer infill (correct width and offset from the wall)
[image]

Next infill layer that will fill three layers. See the centerline offset from the wall due to wrong line thickness.
[image]

smartavionics, 2019-09-18:
I think this ancient PR fixes this issue https://github.com/Ultimaker/CuraEngine/pull/1003

Ghostkeeper, 2019-09-19:
Yes, I agree, it's a duplicate of https://github.com/Ultimaker/Cura/issues/5345.

closed 2019-09-19 by Ghostkeeper