Ultimaker/CuraEngine #1139: memory leak in infill.cpp:640

BrentHuang, 2019-09-25:
**Application Version**

4.3.0
**Platform**

linux
**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

memory leak in infill.cpp:640
            InfillLineSegment* new_segment = new InfillLineSegment(unrotated_first, first.vertex_index, first.polygon_index, unrotated_second, second.vertex_index, second.polygon_index);
            //Put the same line segment in the data structure twice: Once for each of the polygon line segment that it crosses.
            crossings_on_line[first.polygon_index][first.vertex_index].push_back(new_segment);
            crossings_on_line[second.polygon_index][second.vertex_index].push_back(new_segment);

Ghostkeeper, 2019-09-26:
We knew about this one but don't consider it a pressing problem. The memory leaks in CuraEngine are short-lived, because CuraEngine exists as soon as the slice is complete. The operating system cleans it all up then. Pragmatically there is no real issue here, though it will temporarily allocate more memory than necessary so there would be a small performance gain from cleaning this up.

To fix this, I think we need to look towards using smart pointers for these. Either that or deleting the objects on the destructor of the `Infill` class. But we're not giving this any priority right now.

BrentHuang, 2019-09-26:
Thank you, it is very easy to solve this. I use a vector to record the allocated segment and traverse the vector release in the Infill destructor.

Ghostkeeper, 2019-09-27:
They are already in a vector though. In `crossings_on_line`.