Ultimaker/CuraEngine #1140: Unrecognized data type in JSON setting

BrentHuang, 2019-09-25:
**Application Version**

4.3.0

**Platform**

linux

[WARNING] Unrecognized data type in JSON setting, name: machine_disallowed_areas, value: []
[WARNING] Unrecognized data type in JSON setting, name: nozzle_disallowed_areas, value: []
[WARNING] Unrecognized data type in JSON setting, name: machine_head_polygon, value: [[-1,1],[-1,-1],[1,-1],[1,1]]
[WARNING] Unrecognized data type in JSON setting, name: machine_head_with_fans_polygon, value: [[-20,10],[10,10],[10,-10],[-20,-10]]
...
[WARNING] Unrecognized data type in JSON setting, name: machine_head_polygon, value: [[-26,34],[-26,-32],[22,-32],[22,34]]
[WARNING] Unrecognized data type in JSON setting, name: machine_head_with_fans_polygon, value: [[-26,34],[-26,-32],[32,-32],[32,34]]

Ghostkeeper, 2019-09-26:
The warnings about the polygons are sort of expected. CuraEngine can't read those setting values but it doesn't need to read them either. We're limited here by the parsing of RapidJSON to detect the data type in the setting.

I'm not able to reproduce the `machine_max_feedrate_x` and `gantry_height` warnings. Are you sure you're using the most recent versions of both fdmprinter.def.json and CuraEngine?

BrentHuang, 2019-09-26:
Yes, I am using the latest 4.3.0 tag, I have a lot of warning here.
The complete log is as follows (I added some logs)

----------load json file: /home/hgc/workspace/gerrit/CuraEngine/resources/definitions/creality_cr10.def.json
----------load json file: /home/hgc/workspace/gerrit/CuraEngine/resources/definitions/creality_base.def.json
----------load json file: /home/hgc/workspace/gerrit/CuraEngine/resources/definitions/fdmprinter.def.json
----------load json file: /home/hgc/workspace/gerrit/CuraEngine/resources/definitions/fdmextruder.def.json
[WARNING] Unrecognized data type in JSON setting, name: machine_disallowed_areas, value: []
[WARNING] Unrecognized data type in JSON setting, name: nozzle_disallowed_areas, value: []
[WARNING] Unrecognized data type in JSON setting, name: machine_head_polygon, value: [[-1,1],[-1,-1],[1,-1],[1,1]]
[WARNING] Unrecognized data type in JSON setting, name: machine_head_with_fans_polygon, value: [[-20,10],[10,10],[10,-10],[-20,-10]]
----------load json file: /home/hgc/workspace/gerrit/CuraEngine/resources/definitions/creality_base_extruder_0.def.json
----------load json file: /home/hgc/workspace/gerrit/CuraEngine/resources/definitions/fdmextruder.def.json
[WARNING] JSON setting machine_max_feedrate_x has no default_value!
[WARNING] JSON setting machine_max_feedrate_y has no default_value!
[WARNING] JSON setting machine_max_feedrate_z has no default_value!
[WARNING] JSON setting machine_max_feedrate_e has no default_value!
[WARNING] JSON setting machine_max_acceleration_x has no default_value!
[WARNING] JSON setting machine_max_acceleration_y has no default_value!
[WARNING] JSON setting machine_max_acceleration_z has no default_value!
[WARNING] JSON setting machine_max_acceleration_e has no default_value!
[WARNING] JSON setting machine_acceleration has no default_value!
[WARNING] JSON setting machine_max_jerk_xy has no default_value!
[WARNING] JSON setting machine_max_jerk_z has no default_value!
[WARNING] JSON setting machine_max_jerk_e has no default_value!
[WARNING] JSON setting acceleration_print has no default_value!
[WARNING] JSON setting acceleration_travel has no default_value!
[WARNING] JSON setting acceleration_travel_layer_0 has no default_value!
[WARNING] JSON setting acceleration_roofing has no default_value!
[WARNING] JSON setting jerk_print has no default_value!
[WARNING] JSON setting jerk_travel has no default_value!
[WARNING] JSON setting jerk_travel_layer_0 has no default_value!
[WARNING] JSON setting acceleration_enabled has no default_value!
[WARNING] JSON setting jerk_enabled has no default_value!
[WARNING] JSON setting speed_print has no default_value!
[WARNING] JSON setting speed_infill has no default_value!
[WARNING] JSON setting speed_wall has no default_value!
[WARNING] JSON setting speed_wall_0 has no default_value!
[WARNING] JSON setting speed_wall_x has no default_value!
[WARNING] JSON setting speed_topbottom has no default_value!
[WARNING] JSON setting speed_roofing has no default_value!
[WARNING] JSON setting speed_travel has no default_value!
[WARNING] JSON setting speed_layer_0 has no default_value!
[WARNING] JSON setting speed_print_layer_0 has no default_value!
[WARNING] JSON setting speed_travel_layer_0 has no default_value!
[WARNING] JSON setting speed_prime_tower has no default_value!
[WARNING] JSON setting speed_support has no default_value!
[WARNING] JSON setting speed_support_interface has no default_value!
[WARNING] JSON setting speed_z_hop has no default_value!
[WARNING] JSON setting skirt_brim_speed has no default_value!
[WARNING] JSON setting line_width has no default_value!
[WARNING] JSON setting optimize_wall_printing_order has no default_value!
[WARNING] JSON setting material_initial_print_temperature has no default_value!
[WARNING] JSON setting material_final_print_temperature has no default_value!
[WARNING] JSON setting material_flow has no default_value!
[WARNING] JSON setting travel_compensate_overlapping_walls_0_enabled has no default_value!
[WARNING] JSON setting z_seam_type has no default_value!
[WARNING] JSON setting z_seam_corner has no default_value!
[WARNING] JSON setting infill_sparse_density has no default_value!
[WARNING] JSON setting infill_pattern has no default_value!
[WARNING] JSON setting infill_before_walls has no default_value!
[WARNING] JSON setting infill_overlap has no default_value!
[WARNING] JSON setting skin_overlap has no default_value!
[WARNING] JSON setting infill_wipe_dist has no default_value!
[WARNING] JSON setting wall_0_wipe_dist has no default_value!
[WARNING] JSON setting fill_perimeter_gaps has no default_value!
[WARNING] JSON setting fill_outline_gaps has no default_value!
[WARNING] JSON setting filter_out_tiny_gaps has no default_value!
[WARNING] JSON setting retraction_speed has no default_value!
[WARNING] JSON setting retraction_retract_speed has no default_value!
[... 22 lines cut ...]
[WARNING] JSON setting meshfix_maximum_travel_resolution has no default_value!
[WARNING] JSON setting support_angle has no default_value!
[WARNING] JSON setting support_pattern has no default_value!
[WARNING] JSON setting support_infill_rate has no default_value!
[WARNING] JSON setting support_use_towers has no default_value!
[WARNING] JSON setting support_xy_distance has no default_value!
[WARNING] JSON setting support_xy_distance_overhang has no default_value!
[WARNING] JSON setting support_z_distance has no default_value!
[WARNING] JSON setting support_xy_overrides_z has no default_value!
[... cut ...]
[WARNING] JSON setting top_bottom_thickness has no default_value!
[WARNING] JSON setting wall_thickness has no default_value!
[WARNING] Unrecognized data type in JSON setting, name: machine_head_polygon, value: [[-26,34],[-26,-32],[22,-32],[22,34]]
[WARNING] Unrecognized data type in JSON setting, name: machine_head_with_fans_polygon, value: [[-26,34],[-26,-32],[32,-32],[32,34]]
[WARNING] JSON setting gantry_height has no default_value!
----------load json file /home/hgc/workspace/gerrit/CuraEngine/resources/definitions/creality_cr10.def.json finished

BrentHuang, 2019-09-27:
If I run curaEngine alone(without cura GUI), do I need to parse these types?

Ghostkeeper, 2019-09-27:
The ones where it complains about not being able to parse those types are not a problem. CuraEngine doesn't use those settings so it doesn't really need to parse them.

The rest, with not having a default value, could be a problem. Though maybe it works well for you because you provide different JSON files that supplement each other and it's complaining about the individual files. Is there anything weird about how it slices? Like having zero top/bottom thickness?

OffizerBurns, 2019-12-20:
I'm having pretty much exactly the same issue.

I pass the fdmprinter.def.json file along with creality_ender3.def.json (in my case) in a slice command. In the same directory is fdmextruder.def.json, creality_base_extruder_0.def.json, creality_base.def.json which are needed dependencies. The creality_ender3.def.json file inherits from creality_base.def.json and within the creality_base.def.json it overrides for example the default value of max_feedrate etc. which is set in fdmprinter.def.json. My Output Gcode ist empty. For some reason the values for certain attributes are not red out correctly i guess.
I tryed passing the missing attributs with the command which gave me better, yet no good results because i seem to always miss some and it is kinda dumb to write them all out in the command.
Any Idea why the .def files aren't red correctly ?

Ghostkeeper, 2019-12-23:
Is there anything in `stderr` that could indicate why it's not reading out the JSON files?

OffizerBurns, 2020-01-08:
I dont think so …. 

```
Cura_SteamEngine version master
Copyright (C) 2018 Ultimaker

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
[WARNING] Unrecognized data type in JSON setting machine_disallowed_areas
[WARNING] Unrecognized data type in JSON setting nozzle_disallowed_areas
[WARNING] Unrecognized data type in JSON setting machine_head_polygon
[WARNING] Unrecognized data type in JSON setting machine_head_with_fans_polygon
[WARNING] Unrecognized data type in JSON setting machine_disallowed_areas
[WARNING] Unrecognized data type in JSON setting nozzle_disallowed_areas
[WARNING] Unrecognized data type in JSON setting machine_head_polygon
[WARNING] Unrecognized data type in JSON setting machine_head_with_fans_polygon
[WARNING] JSON setting machine_max_feedrate_x has no default_value!
[WARNING] JSON setting machine_max_feedrate_y has no default_value!
[WARNING] JSON setting machine_max_feedrate_z has no default_value!
[WARNING] JSON setting machine_max_feedrate_e has no default_value!
[WARNING] JSON setting machine_max_acceleration_x has no default_value!
[WARNING] JSON setting machine_max_acceleration_y has no default_value!
[WARNING] JSON setting machine_max_acceleration_z has no default_value!
[WARNING] JSON setting machine_max_acceleration_e has no default_value!
[WARNING] JSON setting machine_acceleration has no default_value!
[WARNING] JSON setting machine_max_jerk_xy has no default_value!
[WARNING] JSON setting machine_max_jerk_z has no default_value!
[WARNING] JSON setting machine_max_jerk_e has no default_value!
[WARNING] JSON setting acceleration_print has no default_value!
[WARNING] JSON setting acceleration_travel has no default_value!
[WARNING] JSON setting acceleration_travel_layer_0 has no default_value!
[WARNING] JSON setting acceleration_roofing has no default_value!
[WARNING] JSON setting jerk_print has no default_value!
[WARNING] JSON setting jerk_travel has no default_value!
[WARNING] JSON setting jerk_travel_layer_0 has no default_value!
[WARNING] JSON setting acceleration_enabled has no default_value!
[WARNING] JSON setting jerk_enabled has no default_value!
[WARNING] JSON setting speed_print has no default_value!
[WARNING] JSON setting speed_infill has no default_value!
[WARNING] JSON setting speed_wall has no default_value!
[WARNING] JSON setting speed_wall_0 has no default_value!
[WARNING] JSON setting speed_wall_x has no default_value!
[WARNING] JSON setting speed_topbottom has no default_value!
[WARNING] JSON setting speed_roofing has no default_value!
[WARNING] JSON setting speed_travel has no default_value!
[WARNING] JSON setting speed_layer_0 has no default_value!
[WARNING] JSON setting speed_print_layer_0 has no default_value!
[WARNING] JSON setting speed_travel_layer_0 has no default_value!
[WARNING] JSON setting speed_prime_tower has no default_value!
[WARNING] JSON setting speed_support has no default_value!
[WARNING] JSON setting speed_support_interface has no default_value!
[WARNING] JSON setting speed_z_hop has no default_value!
[WARNING] JSON setting skirt_brim_speed has no default_value!
[WARNING] JSON setting line_width has no default_value!
[WARNING] JSON setting optimize_wall_printing_order has no default_value!
[WARNING] JSON setting material_initial_print_temperature has no default_value!
[WARNING] JSON setting material_final_print_temperature has no default_value!
[WARNING] JSON setting material_flow has no default_value!
[WARNING] JSON setting travel_compensate_overlapping_walls_0_enabled has no default_value!
[WARNING] JSON setting z_seam_type has no default_value!
[... 269 lines cut ...]
Progress:inset+skin:84:100      0.236065%
Progress:inset+skin:85:100      0.238591%
Progress:inset+skin:86:100      0.241116%
Progress:inset+skin:86:100      0.241116%
Progress:inset+skin:87:100      0.243642%
Progress:inset+skin:88:100      0.246167%
Progress:inset+skin:89:100      0.248693%
Progress:inset+skin:89:100      0.248693%
Progress:inset+skin:90:100      0.251218%
Progress:inset+skin:91:100      0.253744%
Progress:inset+skin:91:100      0.253744%
Progress:inset+skin:92:100      0.256269%
Progress:inset+skin:93:100      0.258795%
Progress:inset+skin:93:100      0.258795%
Progress:inset+skin:94:100      0.261320%
Progress:inset+skin:95:100      0.263846%
Progress:inset+skin:95:100      0.263846%
Progress:inset+skin:96:100      0.266371%
Progress:inset+skin:97:100      0.268897%
Progress:inset+skin:97:100      0.268897%
Progress:inset+skin:98:100      0.271422%
Progress:inset+skin:99:100      0.273948%
Progress:inset+skin:1:1         0.276473%
[ERROR] Trying to retrieve setting with no value given: 'max_feedrate_z_override'
```

Ghostkeeper, 2020-01-14:
Can you try setting the `CURA_ENGINE_SEARCH_PATH` environment variable to include the directories of the JSONs that it must find? Something like:

    CURA_ENGINE_SEARCH_PATH=~/Projects/Cura/resources/definitions ~/Projects/CuraEngine/build/CuraEngine slice -j fdmprinter.def.json -j creality_base.def.json -j creality_ender3.def.json -e0 -j fdmextruder.def.json -j ../extruders/creality_base_extruder_0.def.json -l ~/Projects/SlicerTestModels/basic/cube.stl

OffizerBurns, 2020-01-22:
Still crashing at the same Point: 
`[ERROR] Trying to retrieve setting with no value given: 'max_feedrate_z_override'`
But this kinda works:
`./CuraEngine slice -p -j ./fdmprinter.def.json -j creality_ender3.def.json -s max_feedrate_z_override="0" -s material_flow="100" -l part.stl -o part.gcode`
Source: https://github.com/KrisRoofe/curaengine-dockerfile/issues/2

Ghostkeeper, 2020-01-24:
We're unable to reproduce your problem from here when I run it with a locally compiled CuraEngine. Although I didn't try running that Docker file and investigate what's wrong with it either.

fozbek, 2020-02-16:
@OffizerBurns i have fixed the 'no value given' problem in https://github.com/KrisRoofe/curaengine-dockerfile
You should get correct .def.json file. You can get it from same branch.

OffizerBurns, 2020-02-16:
First of all, @fozbek  thank you very much. 

I the meantime i did build my own version of CuraEngine in a plain ubuntu container which seemed to work just the same as your updated Dockerfile. 

I run into a strange issue with mine and your version. The engine finishes without any errors or exeptions but if I run the gcode through a NC Viewer ( e.g https://ncviewer.com/ or from cura frontend itself ) it seems like the Bottom/Top layers are missing. It might just be a viewer related issue but do you know a way to confirm this besides just printing a part irl ? 

I attached a gcode ( as .txt because it didnt let me upload .gcode) which is created with just the default .def.json files ( fdmprinter and fdmextruder) but it is the same if I add more printer specific .def.json files.
And a picture which shows the gcode model from just the CuraEngine backend on the left aswell as a version of the same part made by the frontend on the right.

[test_printer.txt](https://github.com/Ultimaker/CuraEngine/files/4210522/test_printer.txt)

[image]

Ghostkeeper, 2020-02-25:
I see two issues there and I don't know which one you meant, but both are explainable.

The picture on the left shows that the filament diameter is probably different when it was sliced vs. when the g-code was loaded, since the lines are way too thin. The g-code only contains the distance that the feeder has to move, so how wide your lines are depends on the cross-sectional area of the filament. You probably sliced it for a printer with 1.75mm filament and loaded it in Cura into a printer that has 2.85mm filament.

In the right picture, what you're seeing there along the rim at the top is not actually top skin. It's a piece that is just wide enough to fit 5 walls in some places, and just not wide enough to fit 5 walls in other places. This is because of the tessellation of the circle; the circle is approximated with straight line segments, but the corners of those line segments don't match up in the inner circle with the corners of the outer circle. This leads to the wall being slightly thinner in some places than in others. In the wider areas it manages to fit 5 lines (or actually 6 where one is mostly overlapping with another). In the thinner areas it only fits 4 lines and then fills up the gaps with the yellow gap filling things.

OffizerBurns, 2020-02-26:
Okay i see. Thank you @Ghostkeeper.

The issue was indeed the diameter I sliced and laoded the gcode with where differnet. I thaught the top layers where infill because they had gaps in them. But they were top layers all along (facepalm).

KoularaCroft, 2022-07-18:
@OffizerBurns I am having the same issue with the json files? Did you ever figure out what was wrong?

> 'm having pretty much exactly the same issue.
>
> I pass the fdmprinter.def.json file along with creality_ender3.def.json (in my case) in a slice command. In the same directory is fdmextruder.def.json, [...]