Ultimaker/CuraEngine #1141: how to pass model scale and rotate settings to curaEngine?

BrentHuang, 2019-09-26:
**Application Version**

4.3.0

**Platform**

linux

how to pass model scale and rotate settings to curaEngine? with -s?

Ghostkeeper, 2019-09-26:
Exactly! There are a few settings specifically for the command line debug mode to be able to position and rotate your object:

https://github.com/Ultimaker/Cura/blob/648f167911134ce15ac7b9e44619bfa47589817f/resources/definitions/fdmprinter.def.json#L7592-L7640

While the last setting is called "rotation matrix", it can also be used for scaling. You can apply any linear transformation with that matrix.

closed 2019-09-26 by Ghostkeeper

BrentHuang, 2019-09-26:
Thank you, can you give me some configuration examples, such as scaling 1.5 times, then rotating the model, then panning the model?

Ghostkeeper, 2019-09-27:
Sure. Here is a scaling matrix:

    [[1.5, 0, 0], [0, 1.5, 0], [0, 0, 1.5]]

See also documentation [here](https://en.wikipedia.org/wiki/Linear_transformation#Examples_of_linear_transformation_matrices).

Here is a rotation matrix of 30 degrees around the Z axis (you'd have to calculate the cosines and sines before putting this in a setting):

    [[1, 0, 0], [0, cos(30), -sin(30)], [0, sin(30), cos(30)]]

There is a comprehensive formula for it [here on Wikipedia](https://en.wikipedia.org/wiki/Transformation_matrix#Rotation_2) but a more intuitive explanation would be on [Wolfram](http://demonstrations.wolfram.com/VectorRotationsIn3D/).

Panning (moving) the model is not a linear transformation so it can't be represented by the rotation matrix. Use the settings `mesh_position_x`, `mesh_position_y`, and `mesh_position_z` for that.

BrentHuang, 2019-09-27:
There is only one matrix here. What should I do if I want to scale and rotate together?

Ghostkeeper, 2019-09-27:
You multiply the two matrices with matrix multiplication. The order in which the matrix multiplication happens matters though (whether you rotate first and then scale or scale first and then rotate).

BrentHuang, 2019-10-10:
> Sure. Here is a scaling matrix:
>
> ```
> [[1.5, 0, 0], [0, 1.5, 0], [0, 0, 1.5]]
> ```
>
> See also documentation [...]

[[1, 0, 0], [0, cos(30), -sin(30)], [0, sin(30), cos(30)]] => should be rotate by X axis？not Z axis.

ninovanhooff, 2019-10-10:
@BrentHuang This issue was closed by @Ghostkeeper. 
Please direct your question to a maths- or computer graphics related forum.

https://math.stackexchange.com/
https://medium.com/swlh/understanding-3d-matrix-transforms-with-pixijs-c76da3f8bd8

Ghostkeeper, 2019-10-10:
Unrelated to maths: putting `cos(30)` in there like my example won't work. You'll need to calculate the cosine of your angle yourself because CuraEngine will not evaluate it. Whether you do that in degrees, radians, grads, sextants, etc. doesn't matter as long as your calculator does it correctly.