Ultimaker/CuraEngine #1142: can not run twice

BrentHuang, 2019-09-26:
**Application Version**

4.3.0 

**Platform**

linux

[image]
I add for loop in main, but it can not run twice:
[slice -v -j "/home/hgc/workspace/gerrit/CuraEngine/resources/definitions/creality_cr10.def.json" -l "/home/hgc/workspace/gerrit/CuraEngine/tests/yuanbao.stl" -o "/home/hgc/workspace/gerrit/CuraEngine/test.gcode"]

the error log is:
Starting export...
CuraEngine: /home/hgc/workspace/gerrit/CuraEngine/src/gcodeExport.cpp:1003: void cura::GCodeExport::startExtruder(size_t): Assertion `getCurrentExtrudedVolume() == 0.0 && "Just after an extruder switch we haven't extruded anything yet!"' failed.
11:33:08: The program has unexpectedly finished.

Ghostkeeper, 2019-09-26:
You'd need to reset the `FffProcessor` instance in order to run the application multiple times. The `run` function has side effects on FffProcessor's fields and doesn't clean up after itself, since cleaning up after itself would only negatively impact performance for the Cura users. It wouldn't have any advantage for anyone that doesn't modify CuraEngine's source code.

I've documented the requirement now here: 8833877c03d72135e695dd4804db7b39bdc40f7e.

BrentHuang, 2019-09-26:
Thank you, I understand the problem of static variables, but you can try clean the context, not difficult.

Ghostkeeper, 2019-09-27:
It's not difficult, no. That's not the problem. It's just not a good change. As I said, deallocating everything induces a performance hit that we don't want to take. There is no reason for us to support this but there is a reason for us to not support this.

closed 2019-09-27 by Ghostkeeper