Ultimaker/CuraEngine #1147: [ERROR] Trying to retrieve setting with no value given: 'machine_extruder_cooling_fan_number'

Yzhao1125, 2019-09-30:
The command line:
./build/CuraEngine slice -v -j ../Cura-master/resources/definitions/fdmprinter.def.json -o "output/test.gcode" -e1 -s infill_line_distance=0 -e0 -l "../stlmodel/bicycle.stl" 

**Application Version**

4.3.0

**Platform**

ubuntu 18.04
Cura 4.3

Actual result:
Starting slice...
Slicing model...
slice of mesh took 0.262 seconds
slice make polygons took 0.036 seconds
Progress: slice accomplished in 0.299s
Starting layerparts...
Progress: layerparts accomplished in 0.060s
Starting inset+skin...
Layer count: 1381
Progress: inset+skin accomplished in 0.209s
Starting support...
Processing platform adhesion
Progress: support accomplished in 0.435s
Starting export...
[ERROR] Trying to retrieve setting with no value given: 'machine_extruder_cooling_fan_number'

Respect result:
gcode file

Could anyone know the reason of the error? Thank you!

smartavionics, 2019-09-30:
That error typically occurs when your CuraEngine and Cura repos are not in sync. That setting definitely exists within the current master branch.

Yzhao1125, 2019-09-30:
Thank you, Smartavionics, is it what you mean the incompatible of the versions of cura and curaEngine? I am new to 3D printing, could you tell me how to solve this error?

smartavionics, 2019-09-30:
I do not know how you have obtained your CuraEngine and fdmprinter.def.json. If you have built the CuraEngine using the source code from github make sure you have the latest source by doing a "git pull" . The same applies to the fdmprinter.def.json file that is in the Cura repo, do a git pull on that as well to obtain the latest version.

Yzhao1125, 2019-09-30:
Thank you very much. I obtain them by doing a "git clone", and built CuraEngine with source code. is it supposed to be the latest version? Now, I am downloading the latest version from the releases list and then try again

Yzhao1125, 2019-09-30:
I download CuraEngine-4.3.0.tar.gz and Cura-4.3.0.tar.gz, and cmake the CuraEngine. With the above command, the error still happen.

smartavionics, 2019-09-30:
Are you using the fdmprinter.def.json file from Cura-4.3.0.tar.gz?

Yzhao1125, 2019-09-30:
Yes, and I find that there isn't machine_extruder_cooling_fan_number in fdmprinter.def.json at Cura/resources/definitions/, is it the reason why the error happens?

smartavionics, 2019-09-30:
Well, that setting is definitely defined in the 4.3.0 version of that file...
```
~/3dp/cura/Cura.git/build[(4.3.0)]$ git grep machine_extruder_cooling_fan_number ../resources/definitions/fdmextruder.def.json
../resources/definitions/fdmextruder.def.json:               "machine_extruder_cooling_fan_number":

```

Yzhao1125, 2019-09-30:
Yes, I find it, but it seems that my CuraEngine doesn't find it. Should I set CURA_ENGINE_SEARCH_PATH??
CURA_ENGINE_SEARCH_PATH=/path/to/Cura/resources/definitions:/user/defined/path
but how to set the /user/defined/path?

Yzhao1125, 2019-09-30:
I use both fdmprinter.def.json and fdmextruder.def.json to run the command, and the gcode is generated successfully. Thank you very much for your time and help!

smartavionics, 2019-09-30:
OK. If you're happy now, please close this issue. Bye!

closed 2019-09-30 by Yzhao1125

Ghostkeeper, 2019-09-30:
The setting is indeed not defined in fdmprinter.def.json but in fdmextruder.def.json.

gpantalos, 2022-03-08:
> I use both fdmprinter.def.json and fdmextruder.def.json to run the command, and the gcode is generated successfully. Thank you very much for your time and help!

Hi, how did you fix this issue exactly? Can you display the command?

Ghostkeeper, 2022-03-20:
The issue is that it can't find `fdmextruder.def.json`. It finds that either by the current working directory, or by being provided it via `-j`.

So probably the fix was to add both `fdmextruder` and `fdmprinter`, like: `-j /path/to/fdmprinter.def.json -j /path/to/fdmextruder.def.json`.