Ultimaker/CuraEngine #1149: cura 4.3 generates wrong code

elsbiet, 2019-10-10:
**Application Version**
cura 4.3

**Platform**
linux 10 (buster)

**Qt**

**PyQt**

**Display Driver**
??

**Steps to Reproduce**
download .stl from https://www.thingiverse.com/thing:763208?fbclid=IwAR0aOdXwGgXMUH0orhYWRzzP66mrpG_IHL3XAP08XjC7Y7AiUifXDlmO3iA , slize it and compare the result with .stl

**Actual Results**
slizer has added an extra roof including support for it (if support was enabled)

**Expected results**
slizer should not modify the model (except suppor and adhesion)

**Additional Information**

rburema, 2019-10-11:
Right, seems like the surface-normals on that model are the 'wrong' way around. (That's why you some faces show up as red in the preview mode, even though they're not overhang.) Cura _should_ be able to deal with this a bit more gracefully, but it's a known problem.

If you're using the frontend (that is, Cura itself instead of _just_ the plain CuraEngine), then the best workaround known is to download the 'Mesh Tools' plugin from the Marketplace (the button for that is in the upper right). Then, after that is installed and Cura has been restarted, (select the model and) go to 'Extensions' > 'Mesh Tool' > 'Fix model normals'. I tried this and your model slices as normal then.

elsbiet, 2019-10-14:
thanks for the workaround. t think, i can live with it (for a while).

closed 2019-10-14 by elsbiet