Ultimaker/CuraEngine #1151: VeryFine MaxResolution (<0.1mm) results in jagged artifacts and worse slicing

daT4v1s, 2019-10-20:
**Application Version**

Cura 4.3.0 , Official Build

**Platform**

Linux Ubuntu 18.04 LTS , Ultimaker 2+

**Display Driver**

Intel Graphics

**Steps to Reproduce**

set max resolution to LessThan 0.1 mm
slice print
view tool-path result in PreviewPane

**Actual Results**

the lines look tangles,and coarser
i thought a lower number meant a finer result, 
but if was simply coarser the artifacts are expected to be regular, but are jagged here
and also there are occasional spurious blotches

is this due to code being emitted being finder than what the printer can represent?
(heard of 0.04mm but haven't verified this)

also the printing times changed irregularly 
 for other prints when tweaking this value 
(more than 10 minutes at some values for a 2-4 hours print)

**Expected results**

lines look smoother when using a finer resolution,
print paths follow the curve, remain smooth,
no spurious blobs around the print

**Additional Information**

this is still the case after disabling/resetting various settings, except for the resolution related

various resolution setting sweep, starting from 0.01mm (supposedly finest)
notice the random blob @ 0.03mm
[image]

- this became more noticeable when using a finer mesh tessellation (relevant of not?)

daT4v1s, 2019-10-20:
oops - maybe I should have filed this 
in https://www.github.com/Ultimaker/Cura/issues
how shall i move this?

Ghostkeeper, 2019-10-21:
That's okay. Maybe you can provide a project file that reproduces the issue? That's via File -> Save... (**Not** the save button in the bottom left.) I've tried this with a high resolution cylinder but wasn't able to get that effect.

One thing you could try as a test is to disable wall overlap compensation.

daT4v1s, 2019-10-23:
the .3mf cura-file is 34mb, (>10MB attachment limit)
the artifacts are still there despite turning off OverlapCompensation 
[image]

Ghostkeeper, 2019-10-23:
You could upload it elsewhere (WeTransfer?) and post a link here.

I tried to reproduce it by loading a 700MB sphere with 1cm diameter into Cura, the highest resolution model I have. But the surface was smooth.

daT4v1s, 2019-10-24:
[MaxRes bug demo, (AutoDropModelsOff) ZeroHorizontalExpansion .3mf](https://framadrop.org/r/yeNxCsk8Tx#bUIpiUSSSNMMCGAlNYosGrLn/crsIpKTjVwXT9j1xQM=)

Ghostkeeper, 2019-10-25:
Got it. I was thinking maybe that it had something to do with Compensate Wall Overlaps but it doesn't. It seems to be reducing the resolution way too much. I guess 0.01mm is getting somewhat near the slicing resolution of 0.001mm so maybe we're seeing some rounding errors. I'll put up a notice to discuss with the rest of the team.

smartavionics, 2019-10-26:
Hi @daT4v1s , i have a possible fix that you could try if you are willing to install one of my development builds which you can find at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. I will build a release containing the fix tomorrow (Sunday).

I am still seeing some weird roughness in the wall at one end of your model which is odd as the rest of it looks good.

[image]

smartavionics, 2019-10-27:
Hello @daT4v1s , I have now uploaded a new Linux build (1027) if you wish to try it. Please note that in this build, the resolution settings have been moved to the Mesh Fixes section.

I haven't been able to fix the roughness that occurs in one corner of your model. Something weird's going on.

smartavionics, 2019-10-27:
I think that rough region is being created by Cura due to the mesh being less uniform in that region. Firstly, here's an image of the mesh on another portion of the model's outline that slices without the roughness...

[image]

Notice how the triangles on the edge are pretty uniform all the way around that curve.

Now compare that with this image which is the edge that gets sliced all rough...

[image]

It looks very different, doesn't it? Much more random.

Obviously, Cura should cope with that less uniform mesh but at the moment, it doesn't. I'm not familiar with that part of the slicer but I'm guessing the problem lies in the code that takes the mesh and forms the basic polygons?

smartavionics, 2019-10-28:
I reduced the mesh down to 100000 triangles and the rough region has gone. That part of the mesh now looks like this...

[image]

I don't intend to spend any more time on this as now we know that the problem can be avoided.

daT4v1s, 2019-10-30:
@smartavionics  i got your .appimage build to work, only after extracting it;
[image]

also may i ask what program you you to view/edit the mesh, i don't recognise it

also for the part that is rendered rough, whatever the cause,
 the problem is not in the normals of the mesh face 
[image]

smartavionics, 2019-10-30:
>  but typing numbers: numpad does not work, main keyboard does

Sorry, I never use the numpad but I'll look at that.

To view the mesh (as shown in my images above), I use gmsh which I build from the their git source.

I reduced the mesh size using meshlab which I installed from https://launchpad.net/~zarquon42/+archive/ubuntu/meshlab

> also for the part that is rendered rough, whatever the cause,
the problem is not in the normals of the mesh face

I wonder if it is simply the density of the triangles in that region that is causing confusion. Other regions of that sloping edge that goes around your model look quite different.

daT4v1s, 2019-11-01:
the fillet rounds did improve @ fine-resolution
an updated version of the file had different scaling, and surprisingly the 

> weird roughness

was gone at different scaling (0.8% vs ~0.72% shrinkage compensation)
slight changes result in dramatic differences in the wrinkles at the mesh

[image]

at 1.0055× scaling also had this random blotch but after closing and re-opening (not saving) and redo-ing the scaling, i wasn't able to reproduce it
[image] 

is this fuzzing cura? it sometimes hangs during slicing instead of eventually finishing

Ghostkeeper, 2019-11-04:
> Does repeatedly scaling the object within cura introduce rounding errors into the mesh, esp if the mesh is dense (fine spacing between vertices)?

There are two rounding errors that could occur:
1. Floating-point rounding in the scaling matrix. These rounding errors propagate (so repeatedly scaling will increase the error). However the errors are on the order of 0.00000000000002% (10^-16).
2. Rounding to micrometres in CuraEngine. These rounding errors don't propagate, so repeatedly scaling will not increase the error.

Virtox, 2020-03-11:
As my bug report was diverted here, for me it goes awry when max resolution goes below 0.5mm already.
Even with a clean an simple mesh, I have posted a file and images here: 
https://github.com/Ultimaker/Cura/issues/7245

Is it possible to turn of simplification completely some way until it is fixed, at least in surface mode or spiral vase?
I usually have very clean manifold meshes that should be relatively easy to turn into paths that follow the mesh points exactly.

Ghostkeeper, 2020-03-17:
Currently, no. I've ran into that problem myself too. When I need a test that is unmodified by the simplification and has small segments, I'd need to either disable it in CuraEngine's source code (which requires a from-source build of CuraEngine) or I'd need to use the SVGToolpathReader (which is limited in many ways, but sufficient for some tests).

There is no real workaround for this bug, which makes it all the more stringent.