Ultimaker/CuraEngine #1157: CuraEngine from command line dose not support multiple extruders

mirovskyi, 2019-11-07:
**Application Version**
4.4

**Platform**
Linux

**Steps to Reproduce**
$ ./CuraEngine slice -v -j ultimaker3.def.json -o output.gcode -e0 -l T0.stl -e1 -l T1.stl

**Actual Results**
output.gcode has only T0 and does not switch to second T1 extruder

**Expected results**
output.gcode has "T1" line for second model T1.stl

**Additional Information**
All profiles I used are original without any changes.

Ghostkeeper, 2019-11-07:
Yup, seems to be broken. It does load settings for the correct extruder but doesn't load the model. I tried a quick fix but it needs to be debugged properly.

We don't use the debug mode of CuraEngine that often with multiple extruders since it makes things more complex, so I don't know how much force we could put behind this bug...

nallath, 2019-11-08:
The following code will indeed make everything printed by 1 extruder:
` ./CuraEngine slice -v -j fdmprinter.def.json -o output.gcode -e0 -l T0.stl -j fdmextruder.def.json -e1 -j fdmextruder.def.json -l T1.stl`
But if you do:
The following code will indeed make everything printed by 1 extruder:
` ./CuraEngine slice -v -j fdmprinter.def.json -o output.gcode -e0 -l T0.stl -j fdmextruder.def.json -e1 -j fdmextruder.def.json -l T1.stl -s extruder_nr="1"` 
The second object (T1) will be printed by the second extruder.

The reason for this is because the same extruder definition is used for both extruders, which defines the extruder_nr to be 0. So for both models, the engine gets the instruction to print it with extruder 0.

So you can either ensure that the extruder definitions that you give it set the correct value or you can manually change it.

closed 2019-11-08 by nallath

Ghostkeeper, 2019-12-03:
Maybe this is something we need to automate. If you put `-e1` then you might reasonably be expecting `extruder_nr` to be set correctly.

PR #1161 (2019-12-31): Automatically set the extruder number when using command line
So that you don't also need to specify extruder_nr. Makes things easier to type.

Fixes #1157.
commit 8233a320 (2019-12-31): Automatically set the extruder number when using command line

So that you don't also need to specify extruder_nr. Makes things easier to type.

Fixes #1157.