Ultimaker/CuraEngine #1159: Cannot slice stl model global settings missing

averter, 2019-11-20:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

averter, 2019-11-20:
I've also tried to use the aforementioned settings files for the Cura 3.1.0 branch, but the result is the same. I've seen that there are other similar issues for this error but I don't quite understand what is wrong (probably because I am new to the Cura project) and how to fix it.

rburema, 2019-11-22:
(Please note that in general, we only support the latest version.)

For I understand correctly that you download the definitions from the master branch and then try to apply them to an older version of the engine? I can tell, because 'max_feedrate_z_override' was removed for 4.2 as far as I can tell (not in the current engine anymore).

You should make sure the version (of the Cura repo) of the definitions and the version of the CuraEngine are the same. If you're trying to run the engine at 3.1, you should get the definitions of Cura 3.1

averter, 2019-11-25:
Thanks @rburema. I did use Cura 3.1 definitions and now it is working. 
I have one final question though (I hope it makes sense). I'm using CuraEngine mainly to determine the volume and printing time of a given model for certain combination of three parameters. The parameters are the layer height, the infill density and the speed of the print. My question is what settings need to be changed in the settings JSON file based on the above parameters? So far I've associated layer height with a `resolution` setting, infill density with an `infill_sparse_density` and the speed with a `speed_print` setting. Are there other settings beyond these three that are changed when the properties are updated on cura material profiles? Thank you in advance.

rburema, 2019-11-26:
The settings `infill_sparse_density` and `speed_print` are correct, `resolution` is a category though, you probably want `layer_height`.

However! There's a large number of things that depend on those settings. Resolving those dependencies is part of the point of the front-end.

I'd suggest building a plugin for the front-end if you're going to need the same.

averter, 2019-11-26:
Yes, I meant `layer_height``not resolution - thanks for pointing that out.
When you say "a large number of things" do you mean other settings? For example, I think that the `speed_print` also affects these other settings: speed_wall_x, speed_roofing, speed_support_infill, speed_support, speed_prime_tower, speed_infill, speed_wall_0, bridge_skin_material_flow, speed_topbottom. Is there a way to know what these dependencies are to build such a plugin? Should I ask it in the Cura front-end github for example?

Ghostkeeper, 2019-12-04:
> So far I've associated layer height with a resolution setting, infill density with an infill_sparse_density and the speed with a speed_print setting.

Actually if you're calling CuraEngine via the command line you won't be able to use `speed_print` or `infill_sparse_density`, but you need to use the leaf settings, so `speed_infill` and such and `infill_line_distance`. CuraEngine only listens to settings that have no child settings. So yeah you'd need to override all of them.

> Is there a way to know what these dependencies are to build such a plugin? Should I ask it in the Cura front-end github for example?

You can take a look at our [example plug-ins](https://github.com/Ultimaker?utf8=✓&q=UraniumExample) or indeed ask it in the Cura Github page.

closed 2019-12-04 by Ghostkeeper