Ultimaker/CuraEngine #1160: ArcusCommunicationPrivateTest fails on 32-bit

andreasbaumann, 2019-11-26:
`make test` on 32-bit IA32 (for instance Archlinux32) fails with:

```
          8 - ArcusCommunicationPrivateTest (Failed)
```

running `ctest --verbose -R ArcusCommunicationPrivateTest`:

8: Test command: /build/curaengine/src/CuraEngine-4.3.0/build/ArcusCommunicationPrivateTest
8: Test timeout computed to be: 1500
8: [==========] Running 4 tests from 1 test case.
8: [----------] Global test environment set-up.
8: [----------] 4 tests from ArcusCommunicationPrivateTest
8: [ RUN      ] ArcusCommunicationPrivateTest.ReadGlobalSettingsMessage
8: [       OK ] ArcusCommunicationPrivateTest.ReadGlobalSettingsMessage (1 ms)
8: [ RUN      ] ArcusCommunicationPrivateTest.ReadSingleExtruderSettingsMessage
8: [       OK ] ArcusCommunicationPrivateTest.ReadSingleExtruderSettingsMessage (0 ms)
8: [ RUN      ] ArcusCommunicationPrivateTest.ReadMultiExtruderSettingsMessage
8: [       OK ] ArcusCommunicationPrivateTest.ReadMultiExtruderSettingsMessage (0 ms)
8: [ RUN      ] ArcusCommunicationPrivateTest.ReadMeshGroupMessage
8: /build/curaengine/src/CuraEngine-4.3.0/tests/arcus/ArcusCommunicationPrivateTest.cpp:250: Failure
8:       Expected: max_coords[i] - min_coords[i]
8:       Which is: 9900
8: To be equal to: raw_max_coords[i] - raw_min_coords[i]
8:       Which is: 9898
8: /build/curaengine/src/CuraEngine-4.3.0/tests/arcus/ArcusCommunicationPrivateTest.cpp:250: Failure
8:       Expected: max_coords[i] - min_coords[i]
8:       Which is: 9900
8: To be equal to: raw_max_coords[i] - raw_min_coords[i]
8:       Which is: 9898
8: /build/curaengine/src/CuraEngine-4.3.0/tests/arcus/ArcusCommunicationPrivateTest.cpp:250: Failure
8:       Expected: max_coords[i] - min_coords[i]
8:       Which is: 9900
8: To be equal to: raw_max_coords[i] - raw_min_coords[i]
8:       Which is: 9899
8: [  FAILED  ] ArcusCommunicationPrivateTest.ReadMeshGroupMessage (2 ms)
8: [----------] 4 tests from ArcusCommunicationPrivateTest (3 ms total)
8: 
8: [----------] Global test environment tear-down
8: [==========] 4 tests from 1 test case ran. (3 ms total)
8: [  PASSED  ] 3 tests.
8: [  FAILED  ] 1 test, listed below:
8: [  FAILED  ] ArcusCommunicationPrivateTest.ReadMeshGroupMessage
8: 
8:  1 FAILED TEST
1/1 Test #8: ArcusCommunicationPrivateTest ....***Failed    0.01 sec
```

Ghostkeeper, 2019-12-05:
I cannot reproduce this issue using Ubuntu 19.10 64-bit.

```
ruben@UM-MBA-187:~/Projects/CuraEngine/build$ ctest --verbose -R ArcusCommunicationPrivateTest
UpdateCTestConfiguration  from :/home/ruben/Projects/CuraEngine/build/DartConfiguration.tcl
Parse Config file:/home/ruben/Projects/CuraEngine/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/ruben/Projects/CuraEngine/build/DartConfiguration.tcl
Parse Config file:/home/ruben/Projects/CuraEngine/build/DartConfiguration.tcl
Test project /home/ruben/Projects/CuraEngine/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 9
    Start 9: ArcusCommunicationPrivateTest

9: Test command: /home/ruben/Projects/CuraEngine/build/ArcusCommunicationPrivateTest
9: Test timeout computed to be: 1500
9: [==========] Running 4 tests from 1 test suite.
9: [----------] Global test environment set-up.
9: [----------] 4 tests from ArcusCommunicationPrivateTest
9: [ RUN      ] ArcusCommunicationPrivateTest.ReadGlobalSettingsMessage
9: [       OK ] ArcusCommunicationPrivateTest.ReadGlobalSettingsMessage (1 ms)
9: [ RUN      ] ArcusCommunicationPrivateTest.ReadSingleExtruderSettingsMessage
9: [       OK ] ArcusCommunicationPrivateTest.ReadSingleExtruderSettingsMessage (0 ms)
9: [ RUN      ] ArcusCommunicationPrivateTest.ReadMultiExtruderSettingsMessage
9: [       OK ] ArcusCommunicationPrivateTest.ReadMultiExtruderSettingsMessage (0 ms)
9: [ RUN      ] ArcusCommunicationPrivateTest.ReadMeshGroupMessage
9: [       OK ] ArcusCommunicationPrivateTest.ReadMeshGroupMessage (2 ms)
9: [----------] 4 tests from ArcusCommunicationPrivateTest (3 ms total)
9: 
9: [----------] Global test environment tear-down
9: [==========] 4 tests from 1 test suite ran. (4 ms total)
9: [  PASSED  ] 4 tests.
1/1 Test #9: ArcusCommunicationPrivateTest ....   Passed    0.02 sec

The following tests passed:
	ArcusCommunicationPrivateTest

100% tests passed, 0 tests failed out of 1

Total Test time (real) =   0.02 sec
```

andreasbaumann, 2019-12-05:
Yes, because it's a rounding error which happens on non-64-bit. I also have no problem on
Archlinux 64-bit.
Some numbers are 9998 or 9999 on 32-bit, which indicates rounding issues before multiplication.
I can dig in deeper, if you like as I'm aware 32-bit is nowadays a little bit rarer than it used to be. :-)

Ghostkeeper, 2019-12-06:
Yeah, we dropped 32-bit support after version 2.4 (3 years ago).

closed 2019-12-06 by Ghostkeeper