Ultimaker/CuraEngine #1164: Bug: Layertime is miscalculated at G92E0

TimoErdmann, 2019-12-06:
**Application Version**

Cura 

**Expected results**
Layertime should be calculated correct for every layer

**Additional Information**

If the extruded Material Axis E is reset with a G92 the calculated layertime is wrong. It differs over 200% between the layer with the reset and the following or previous one. 

You can check this by making a barrel or a stl with identical layers and observing the calculated print duration.

Ghostkeeper, 2019-12-09:
Yup, this reproduces for me as well. I sliced a tall cylinder, set the infill density to 100% and the seam to back so that every layer but the first few are the same. In the resulting g-code I removed every line except the layer time comment and the G92 E0 commands. This is the result: [test2.zip](https://github.com/Ultimaker/CuraEngine/files/3939887/test2.zip)

As can be seen:

```
;TIME_ELAPSED:10620.765528
;TIME_ELAPSED:10655.062670
;TIME_ELAPSED:10689.360034
G92 E0
;TIME_ELAPSED:10758.158305
;TIME_ELAPSED:10792.455488
;TIME_ELAPSED:10826.752740
```

The difference is about 35s every layer. Except the layer that has the `G92` in it, which is about 80s. So it seems that there is indeed a problem with the time estimation there; the layer in layer view appears identical.

ninovanhooff, 2019-12-16:
Devs: I created https://jira.ultimaker.com/browse/CURA-7053
@rburema I believe you were planning on adding a ticket to Jira, please confirm we don't have a duplicate now.

rburema, 2019-12-17:
@ninovanhooff I wanted to investigate if it would be an easy fix first, I did that, but then something else must've come up. I still have the sticky, so your ticket must be the only one.

smartavionics, 2020-01-05:
Perhaps something like the following is needed? (untested, YMMV)

````
diff --git a/src/gcodeExport.cpp b/src/gcodeExport.cpp
index 65bd713a..d0bd0b65 100644
--- a/src/gcodeExport.cpp
+++ b/src/gcodeExport.cpp
@@ -598,6 +598,7 @@ void GCodeExport::resetExtrusionValue()
     }
     current_e_value = 0.0;
     extruder_attr[current_extruder].retraction_e_amount_at_e_start = extruder_attr[current_extruder].retraction_e_amount_current;
+    estimateCalculator.resetExtrusionValue();
 }
 
 void GCodeExport::writeDelay(const Duration& time_amount)
diff --git a/src/timeEstimate.cpp b/src/timeEstimate.cpp
index 5dc060c6..028cfb7d 100644
--- a/src/timeEstimate.cpp
+++ b/src/timeEstimate.cpp
@@ -60,6 +60,11 @@ void TimeEstimateCalculator::reset()
     blocks.clear();
 }
 
+void TimeEstimateCalculator::resetExtrusionValue()
+{
+    currentPosition.axis[E_AXIS] = 0.0;
+}
+
 // Calculates the maximum allowable speed at this point when you must be able to reach target_velocity using the 
 // acceleration within the allotted distance.
 static inline Velocity max_allowable_speed(const Acceleration& acceleration, const Velocity& target_velocity, double distance)
diff --git a/src/timeEstimate.h b/src/timeEstimate.h
index a9aa586d..c09619d7 100644
--- a/src/timeEstimate.h
+++ b/src/timeEstimate.h
@@ -96,6 +96,7 @@ public:
     void setMaxXyJerk(const Velocity& jerk); //!< Set the max xy jerk to \p jerk
 
     void reset();
+    void resetExtrusionValue();
     
     std::vector<Duration> calculate();
 private:
````

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.