Ultimaker/CuraEngine #1165: Bug: supports are hanging in the air

extesy, 2019-12-08:
**Application Version**
Master 12/8/2019 with @smartavionics enhancements

**Steps to Reproduce**
Slice the model with supports enabled

**Actual Results**
There's one line of supports in the middle of screenshot - they are hanging in the air, the model is not even close:
[image]

**Expected results**
No supports that start in the air.

Ghostkeeper, 2019-12-09:
How can we reproduce this problem? Your reproduction steps indicate to "slice the model" but with several models I didn't find a case like this.

extesy, 2019-12-09:
@Ghostkeeper I actually have seen this bug quite often. What is the best way to attach a repro case? I assume it should include both the model and configs. Does `3mf` format include both? Is there any good way to export values for all settings that I use?

rburema, 2019-12-10:
> Does 3mf format include both?

Yes, at least when you save (as opposed to export). To attach it to a github comment, you have to zip it (or rename) as github is very particular about what formats it accepts.

> By the way, I have created custom definition+quality configs for my printer so if I simply export profile it will probably only include deltas from that definition which won't be very useful.

In that case, please also attach those definitions.

extesy, 2019-12-14:
Here is a project: [dipper.zip](https://github.com/Ultimaker/CuraEngine/files/3964097/dipper.zip). I've looked into 3mf and it's actually an archive itself and it contains cfg overrides I've done as well. Hopefully it's enough for repro. I've sliced it using 20191214 build from https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0 from @smartavionics page.

Surprisingly, I wasn't able to reproduce _exactly_ the same bug, perhaps I had tweaked some settings since then, but there are still plenty of similar problems, for example: 
[image]

Or this: 
[image]

Or this:
[image]

It apparently uses some weird criteria to determine if there's anything underneath support. Maybe it uses information from some _other_ support mesh for optimization purposes and incorrectly assumes that it applies for the _current_ mesh? Just a guess...

smartavionics, 2019-12-14:
I tried loading your project file but because you are using a printer definition that is unknown to my Cura, it only loads the model and no settings. That's pretty useless. I sliced the model using the default support settings for the CR10 and it looks OK at a quick glance.

[image]

Also, your images above are very hard to understand, I cannot find the equivalent views looking at my model because you have zoomed in so much.

Sorry for not being very helpful.

extesy, 2019-12-14:
Well, the configs are supposed to be inside 3mf project but I guess Cura only saves them but doesn't load them back. So I have archived my entire `C:\Users\olega\AppData\Roaming\cura\master\` directory:
[all-configs.zip](https://github.com/Ultimaker/CuraEngine/files/3964224/all-configs.zip) - please let me know if this works now.

smartavionics, 2019-12-14:
Thanks for the other file, I have managed to load your project. I'm not really knowledgeable about support so I don't have any great insights. I increased the Support Horizontal Expansion to 2 and reduced the Minimum Support Area to 1 and the results don't look too bad. I can't really see the big gaps between the interface and the model as you show in your pictures above. 
Personally, I wouldn't use just lines support without an enclosing wall as I would be concerned that they would simply fall over.

[image]

smartavionics, 2019-12-14:
BTW my Cura builds do have a small difference to the UM builds with regard to support which is that it recognizes when support is above a sloping surface and so uses the min x/y distance (see https://github.com/Ultimaker/CuraEngine/pull/1115) for details.

If your problem is not related to my Cura builds as such, it would probably be better that you use an UM release so that the UM devs know what you are using.

ninovanhooff, 2019-12-16:
ultimaker dev here. @extesy please let us know whether the suggested configuration changes work for you in the official Ultimaker builds (either 4.4 or latest master).

Ghostkeeper, 2019-12-16:
> Or this:
[image]

This one seems to be expected; it's the stair steps. So you'd need to set the stair step width and height to 0.

Ghostkeeper, 2019-12-16:
Most of the problems in the screenshots only look bad because it's using Lines support. Lines indeed works pretty badly as well because it tends to miss support parts on the side, and then it appears like the support is doing nothing, and higher up there is support being printed in mid air, while the actual support area is growing towards the side and back. I advise to switch to ZigZag instead since it's very similar to lines but has fewer problems. Here is an article about the advantages and disadvantages of the different infill patterns in my settings guide: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_pattern.md#lines

When you switch to Zigzag (or add an outline) you'll see that there are some parts that are not supporting all the way to the top:
[image]
That's the problem you're seeing in the second screenshot of [this](https://github.com/Ultimaker/CuraEngine/issues/1165#issuecomment-565736014) comment as well. This is because of the Minimum Support Area (a very confusing setting admittedly), so I concur with SmartAvionics that you need to reduce that setting and 1mm² seems fine to me.

Setting the Support Horizontal Expansion to half a line width (0.175mm) or more is usually good to make sure that small details get supported, too.

closed 2019-12-16 by Ghostkeeper