Ultimaker/CuraEngine #1171: Shell before inner & outer walls on initial layer

jancoow, 2019-12-27:
Hi.

Would it be possible to flip the printing order on the initial layer, so the shell gets printed before the inner and outer walls? I'm always having trouble printing small sequences of walls at the initial layer to stick properly to the bed. It would be great if the shell could be printed first, so it has a large area to stick tiny walls to. This would help the adhesion.

Ghostkeeper, 2019-12-31:
Did you try reducing the Small Feature Initial Layer Speed?

jancoow, 2019-12-31:
Yes I did, but it doesn't really help. I mostly have these issues with a big nozzle (0.8mm) and PETG. I've manually flipped the order in the gcode on the first layer and I was finally able to print a specific model, which didn't succeed before. This is because it's first doing the outer walls of a bunch of 3mm holes, where the path is way to short to stick properly.

Ghostkeeper, 2020-01-02:
See also https://github.com/Ultimaker/Cura/issues/6734.

fvrmr, 2021-08-18:
Closing this issue, since it is a duplicate of https://github.com/Ultimaker/Cura/issues/2988

closed 2021-08-18 by fvrmr