Ultimaker/CuraEngine #1176: [Bug - Fix Provided] Prime tower skips layers for non T0 tools.

mkudzia84, 2020-01-07:
Current prime tower behaviour doesn't really work as priming is only happening during the tool change.
This means that in most case scenarios (in dual extruder setup) T1 is only primed on even-numbered layers.
In multi extruder situations this is becoming even worse; 

Example scenario for dual extruder setup:

```
layer #1 : T1 primes, T1 prints, T1->T2, T2 primes on the brim, T2 prints
layer #2 : T2 prints, T2->T1, T1 primes (the outer most layer)
layer #3 : T1 prints, T1 primes (the outer most layer of prime towre is always constructed), T1 -> T2.....fail
```
T2 primes, but because there was no material printed in the T2 prime tower band in previous layer - priming will happen in thin air and fail

Situation is even worse in multi extruder configs.

I have developed patch on 4.4.1 branch that changes the behaviour to a full proof one:

The new behavious is:
- Prior to tool change, slicer checks if the current tool has been primed on the current layer, if not it 'de-primes' it into the designated tool band
- After tool change, new tool is primed, as in main branch
- At the end of the layer, slicer checks if there are any Tool specific bands that been primed and deposits material there using the last active tool.

This:
- Ensures that all prime tower bands are printed at all times
- Doesn't add any extra tool changes;
  1) populating of the prime tower bands is done only during tool change [previous][new] tool
  2) end of processLayer; unused tool prime tower bands are printed with the last active tool (no extra tool change)

This guarantees that all used prime tower bands (non zero minimum prime volume) will be populated on every layer.

I have tested it in various configurations on my 4 toolhead E3D tool changer.

The code is in the fork on:
https://github.com/mkudzia84/cura-toolchanger in 4.4 branch

The fix is in this commit:
https://github.com/mkudzia84/cura-toolchanger/commit/28e7f0cb70237a6143398e2f3552ba8ce491fc69

rburema, 2020-01-12:
Thanks for the work put in!

We still have a bit of a backlog for PR's, so it might be a while until we can evaluate this. However, quickly scanning this, the following line stuck out to me:
> I have developed patch on 4.4.1 branch

We _mostly_ only bring out patches for blockers like fatal crashes and similar things. Since 4.4.1 is already out, it's highly unlikely that we continue developing the 4.4 branch (we merge them back into master). In most cases, the master branch should be used.

(Also, I'm curious: What types of failure(s) would you say results from the mid-air priming?)

mkudzia84, 2020-01-13:
Thanks for reponse, I will merge the fix into master branch (need to build the master branch gui on my side first)

In terms of failures(s)
1) 'priming' mid air doesn't equalize the pressure in the nozzle, leading to underextrusion/overextrusion
2) you end up with tons of strings of filament on the build plate
3) sometimes the 'mid air' prime string of filament will get pulled out of the tower permiter and get attached to the perimeter of the print
4) sometimes the 'mid air' prime string of filament gets stuck on the perimeter of the prime tower while the extruder moves out (yay) but because it gets pulled forcibely, some excess material - due to the viscosity, gets pulled out of the nozzle, ending with sligh under extrusion artifact on the print.

From what I checked on the master branch, issue still there persists.
I will try to do a merge tommorow/day after :)

mkudzia84, 2020-01-13:
Ah what I am saying, I developed the fix on 4.4.1 branch (not the master branch) as this was a show-stopped for my multiextruder setup. I have my own build. But issue persists on the master branch so if you can merge it back to the master - perfect 

If not - Can do it myself, I just don't have the build env for Cura/GUI build yet.

Ghostkeeper, 2020-01-13:
Duplicate of https://github.com/Ultimaker/Cura/issues/6909. Please continue discussion there.

closed 2020-01-13 by Ghostkeeper