Ultimaker/CuraEngine #1179: Maybe there is a better way to generate inner perimeter?

b-desconocido, 2020-01-11:
**Application Version**
Cura 4.4.1

**Platform**

Windows 10

**Steps to Reproduce**
Open Cura
Slice a thin round model, example attached

**Actual Results**
The results are horrible: https://youtu.be/0cM-C30_mDY

**Expected results**
Something like Simplify3D outputs for the same model

**Additional Information**

STL:
[shopa_test.zip](https://github.com/Ultimaker/CuraEngine/files/4049650/shopa_test.zip)

Timocop, 2020-01-13:
And how does the print model look? I dont see anything horrible.
If you're working with small models you want to use combing "Not in skin" instead otherwise youll get quite alot of retraction if you use "Within infill".

Also, you should keep away from Zig Zag skin pattern, its buggy and wont infill very small skin gaps. Keep it on Lines pattern, its the same as Zig Zag anyways. The ozzed material will print on the travel path.

Ghostkeeper, 2020-01-14:
> And how does the printed model look? I dont see anything horrible.

You can see it going back and forth in the preview. That's a bad sign because it'll interrupt the flow of material and ooze while moving, creating blobs and vibrations and inconsistent flow.

This is a duplicate of https://github.com/Ultimaker/Cura/issues/6676.

It's the problem of fitting lines in a thin shape where the thickness is not exactly equal everywhere, and is meant to be an exact multiple of the line width before rounding. Reducing the inner wall thickness to something slightly lower than the wall thickness fixes this problem. It could be that Simplify3D happens to have a better line width for your model by default. Simplify3D of course has [the same problem](https://forum.simplify3d.com/viewtopic.php?f=9&t=12614&p=49429), just with different line and wall widths.

closed 2020-01-14 by Ghostkeeper

Timocop, 2020-01-15:
> You can see it going back and forth in the preview. That's a bad sign because it'll interrupt the flow of material and ooze while moving, creating blobs and vibrations and inconsistent flow.

Oh now i see. Wouldnt a higher "Minimum Wall Flow" value fix this too?

Ghostkeeper, 2020-01-16:
No, that will still make it do the travel moves I think.