Ultimaker/CuraEngine #1181: Compiling curaengine on Windows

abtionlove, 2020-01-13:
I used cygwin64 + CodeBlocks on windows to compile curaengine15.04.6. I used cygwin64 to install protobuf-all-3.9.0 without reporting an error. Then I installed Python and libarcus-4.4.1 (but I didn't install python3 Dev and python3 SIP dev, which I don't know how to install) without reporting an error. Finally, the following error occurs when using CodeBlocks to compile and run: how to solve it, thank you very much.
||=== Build: Release in curaengine (compiler: Cygwin GCC) ===|
ld    cannot find -lmingw32
ld    cannot find -lSDLmain
ld    cannot find -lSDL
ld    cannot find -lmingw32
ld    cannot find -lSDL2main
ld    cannot find -lSDL2
ld    cannot find -lsdL2_image
|||error: ld returned 1 exit status|
||=== Build failed: 8 error(s), 0 warning(s) (0 minute(s), 0 second(s)) ===|

Ghostkeeper, 2020-01-14:
Looks like your linker is unable to find MinGW32 and SDL. Make sure that those dependencies are on your path. We solved this in 15.06 by switching to CMake.

closed 2020-01-14 by Ghostkeeper

abtionlove, 2020-01-15:
> Looks like your linker is unable to find MinGW32 and SDL. Make sure that those dependencies are on your path. We solved this in 15.06 by switching to CMake.

How to switch from 15.06 to cmake?

Ghostkeeper, 2020-01-15:
You can switch to CMake by installing CMake from [here](https://cmake.org/) and using it to compile a more modern branch of CuraEngine. Of course that also requires you to use a more modern version of the front-end.

abtionlove, 2020-01-16:
> You can switch to CMake by installing CMake from here and using it to compile a more modern branch of CuraEngine. Of course that also requires you to use a more modern version of the front-end.

OK， thank you very much.