Ultimaker/CuraEngine #1182: Cannot Build CuraEngine 3.3.0 on RHEL7

mig8447, 2020-01-15:
**Application Version**
3.3.0

**Platform**
Oracle Enterprise Linux 7.7

**Steps to Reproduce**
Follow the installation instructions up to the point of CuraEngine's `make`

**Actual Results**
```
[  2%] Built target clipper
[  3%] Building CXX object CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o
In file included from /tmp/CuraEngine/src/pathPlanning/Comb.h:11:0,
                 from /tmp/CuraEngine/src/LayerPlan.h:8,
                 from /tmp/CuraEngine/src/FffGcodeWriter.h:15,
                 from /tmp/CuraEngine/src/FffProcessor.h:5,
                 from /tmp/CuraEngine/src/commandSocket.cpp:7:
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(Args ...)’:
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:18: error: expected ‘,’ before ‘...’ token
             [args...]()
                  ^
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:18: error: expected identifier before ‘...’ token
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:21: error: parameter packs not expanded with ‘...’:
             [args...]()
                     ^
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:21: note:         ‘args’
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:36:34: error: expansion pattern ‘args’ contains no argument packs
                 return new T(args...);
                                  ^
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(const std::function<_Res(_ArgTypes ...)>&, Args ...)’:
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:21: error: expected ‘,’ before ‘...’ token
             [f, args...]()
                     ^
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:21: error: expected identifier before ‘...’ token
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:24: error: parameter packs not expanded with ‘...’:
             [f, args...]()
                        ^
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:24: note:         ‘args’
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:54:36: error: expansion pattern ‘args’ contains no argument packs
                 return new T(f(args...));
                                    ^
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(const std::function<T*(Args ...)>&, Args ...)’:
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:21: error: expected ‘,’ before ‘...’ token
             [f, args...]()
                     ^
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:21: error: expected identifier before ‘...’ token
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:24: error: parameter packs not expanded with ‘...’:
             [f, args...]()
                        ^
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:24: note:         ‘args’
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/tmp/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:70:30: error: expansion pattern ‘args’ contains no argument packs
                 return f(args...);
                              ^
make[2]: *** [CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o] Error 1
make[1]: *** [CMakeFiles/_CuraEngine.dir/all] Error 2
make: *** [all] Error 2
```

**Expected results**
A successful build

**Additional Information**
Added all the dependencies, built and installed protobuf and libArcus 3.3.0, tested with g++ 4.8.5, 4.9.2, and 7.3.1 (Latter two from `devtoolset-*` package) and still failing on the same place every time

Ghostkeeper, 2020-01-15:
We're building CuraEngine with g++ 4.9.2 and not getting this error.

Is the error still there when you update to the latest stable (or latest master)?

mig8447, 2020-01-16:
I would need to try out with `master` but I want version 3.3.0 compiled because I have a GCode post-processing script that doesn't work well with later versions

rburema, 2020-01-24:
@mig8447 Seems it doesn't know what to do with [template parameter packs](https://en.cppreference.com/w/cpp/language/parameter_pack). Maybe the C++11 standard isn't enabled? You can do so by adding `-std=c++11` to your compiler arguments.

Or maybe it's related to these bugs, of which one was fixed in 8.0 and the other in 8.2 (of gcc):
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=47226
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=85305

mig8447, 2020-02-09:
Tried with version `8.3.1` from `devtoolset-8` package and it is not working still. Fails with the same error. I found that there are some files that define the -std flag:

```
./CMakeFiles/clipper.dir/flags.make:CXX_FLAGS =  -O2 -DNDEBUG -Wall -static-libstdc++   -std=gnu++11
./CMakeFiles/_CuraEngine.dir/flags.make:CXX_FLAGS =  -O2 -DNDEBUG -Wall -static-libstdc++   -std=gnu++11
./CMakeFiles/CuraEngine.dir/flags.make:CXX_FLAGS =  -O2 -DNDEBUG -Wall -static-libstdc++   -std=gnu++11
```

Changed `gnu++11` to `c++11` and still had no effect will try to create a Dockerfile so the issue can be replicated by someone else. Thanks for the help

smartavionics, 2020-02-09:
On the Pi4 I am building the current CuraEngine source using g++ 8.3.0 without a problem.

Ghostkeeper, 2020-02-12:
Can you try switching over to C++14? You can do that by editing this line:

https://github.com/Ultimaker/CuraEngine/blob/c9b5b85b0c7f21f570429dda10d7d85c53d4f4b7/CMakeLists.txt#L48

Maybe your compiler is somehow interpreting the standard differently than ours.