Ultimaker/CuraEngine #1183: [4.4.1] Walls close to nozzle size are considered thin

DenKoren, 2020-01-18:
**Application Version**
4.4.1

**Platform**
Mac OS X 10.14.6

**Display Driver**
Graphics card: Intel Iris Plus Graphics 640 1536 MB
Have no idea about video driver used. The one available out of the box in Mac OS X for MacBook Pro (13-inch, 2017, Two Thunderbolt 3 ports)

**Steps to Reproduce**
Make the brick with length or width close to nozzle size (e.g. for nozzle 0.4mm create a wall 0.4mm or 0.41mm), then slice it. I used 0.4x5x5 brick for test, also I used big box (100x100x100mm) without top surface with outer walls of 0.8mm width and several partitions inside with width 0.4

**Actual Results**
Walls are printed and considered 'thin' regardless of nozzle size.

**Expected results**
Walls are printed since they are greater or equal to the nozzle size without 'print thin walls' setting

**Additional Information**
At some point engine starts to print walls without 'print thin walls' setting. I believe it is somewhere between 0.41 and 0.45 wall width for 0.4 nozzle size.

DenKoren, 2020-01-18:
[Box-thin-walls_0.4.3mf.zip](https://github.com/Ultimaker/CuraEngine/files/4081657/Box-thin-walls_0.4.3mf.zip)

here is the file with walls of 0.4 width inside the box and 0.8 outer walls

DenKoren, 2020-02-05:
Oh, I missed your reply. Sorry for the long response.

Thanks for noting 'inset' parameter value. I re-checked, and slicer still skips walls of 0.4 even with outer wall inset = 0 and all types of wall width = 0.4. But it starts to print them when I set the 'outer wall inset' to any negative value, even -0.001 is enough.

Ghostkeeper, 2020-02-11:
Yes, that's behaving as expected then. When printing something of exactly 0.4mm thick, this part of the polygon is skipped entirely as to not produce degenerate contours.

closed 2020-02-11 by Ghostkeeper