Ultimaker/CuraEngine #1184: [4.4.1] Crazy wall printing order for grid of thin walls

DenKoren, 2020-01-18:
**Application Version**
Cura 4.4.1

**Platform**
Mac OS X 10.14.6

**Display Driver**
Graphics card: Intel Iris Plus Graphics 640 1536 MB
Have no idea about video driver used. The one available out of the box in Mac OS X for MacBook Pro (13-inch, 2017, Two Thunderbolt 3 ports)

**Steps to Reproduce**
Create an object with grid of walls of width close to nozzle size.
I created the box with partitions inside. It's like 'grid' infill pattern, but it is a part of the model since I want to get the exact step between walls and exact wall size, and they don't need to end with a roof.

**Actual Results**
The slicer slices the model but chooses the very ineffective printing order, increasing printing time and reducing quality. The grid cells are printed 'separately', each cell walls are printed at once, then nozzle goes to the next cell.

**Expected results**
The walls grid can be printed by lines, crossing each other. For example, printer can print all 'horizontal' lines and then print all 'vertical' ones. This will be much faster because of less 'empty' hotend moves and also will allow to use faster speed with the same level of wobbling.

**Additional Information**
I know this behavior might be because of attempt of CuraEngine to prevent wall lines crossing. But for this particular case it is much better to allow them crossing.
If it is true and without this restriction it could print the grid as it does for 'infill' pattern, then the option 'allow wall lines crossing' would be useful in some scenarios.

Thanks.

DenKoren, 2020-01-18:
[BoxWallsGrid_0.5.3mf.zip](https://github.com/Ultimaker/CuraEngine/files/4081694/BoxWallsGrid_0.5.3mf.zip)

Here is the file that causes ineffective walls printing order.

This is also the case to be considered for new (and awesome) 'Optimize wall printing order' setting.

DenKoren, 2020-01-18:
For that particular model I attached it is possible to bypass the limitation by precise tune of grid infill pattern.
I created the solid box of outer model's shell, then set infill lines directions to 0 and 90 degrees, set step to fixed size in millimeters and got print time boost of 1/3 for the same printing speed (yes, I set the infill speed same to the 'walls' value). This saves 1 hour for my model and it is just because of travel time reduction.

Unfortunately, this hack only works for cases when the grid has square shape. If one needs to print rectangular grid with different X and Y size, this hack doesn't work any more.
To cover this case we need to control infill line distance with 2 parameters for 'grid' or 'lines' infills: distance 1 and distance 2 for each direction. If it is hard to teach 'walls' slicer to allow lines overlap, or if it wouldn't help, maybe it will be easier to make more customizable line distance option of infill pattern.

smartavionics, 2020-01-19:
Cura doesn't "see" that model as a grid of lines. Instead, it considers it to be a collection of holes and prints the outline of each hole. That behaviour is fundamental to how Cura operates. I can't see that changing any time soon.

Ghostkeeper, 2020-01-21:
It is forced to start each outline in the sharpest corner for the Z Seam Alignment setting. In your case that is set to Sharpest Corner, so it'll look for the sharpest corner of each square, which apparently ends up being the top left square for each of the squares (probably due to rounding errors in the modeller or the slicer). And so it travels to that corner every single time.

If you set the Z Seam Alignment to "Shortest", it'll be a bit better. It will start each polygon on the corner that is closest to it.

But indeed it still can't print all horizontal lines and all vertical lines continuing. Like Smartavionics said, Cura doesn't see the lines as being continuous because they are not. It actually sees more like this:

```
 __   __   __
|__| |__| |__|
 __   __   __
|__| |__| |__|
 __   __   __
|__| |__| |__|
```

That is, just a collection of boxes. The borders of the boxes happen to be placed very closely together though, but Cura doesn't care since it's still just a bunch of contours.

closed 2020-01-21 by Ghostkeeper

DenKoren, 2020-01-21:
Ok, I see, sounds reasonable.
Thanks for responses and the configuration suggestion.

What about the idea of additional distance parameter for grid infills?

Ghostkeeper, 2020-01-24:
Cura is not a modelling tool, so your use case for it, while easy in this case, is not something we strive for to implement.

You could argue though that varying the distance between lines in different directions is a way to optimise rigidity/strength in a certain direction.
However given the wide variety of infill patterns and the infill angles setting that also allow you to customise this to a great extent, I don't think it would be widely used. It only applies to one specific model, so you'd have to adjust it for every model separately and it can't be put in a default profile. So then only the people who find and understand this setting would adjust it, and only when the current solutions for non-isotropic strength are insufficient. That's why I don't think it's a good idea to add that.

DenKoren, 2020-01-25:
Ok. Thanks for considered response.