Ultimaker/CuraEngine #1188: [4.4] Startup weirdness

sk8nfool, 2020-01-25:
**Application Version**
4.4

**Platform**
Ubuntu 18.04.3 LTS

**Qt**
Whatever is in Ultimaker_Cura-4.4.1.AppImage

**PyQt**
Ultimaker_Cura-4.4.1.AppImage

**Display Driver**
Not sure

**Steps to Reproduce**
See below

**Actual Results**
See below

**Expected results**
See Below

**Additional Information**
This is a bit long so ...
I'm running Ubuntu 18.04 on a Dell Optiplex 790. I have reset Cura my removing both
    .local/share/cura/4.4 (also stderr.log and stdout.log in this folder)
    .config/cura/4.4
When I then start Ultimaker_Cura-4.4.1.AppImage by entering in a terminal
    Downloads/Cura/Ultimaker_Cura-4.4.1.AppImage
the splash displays and a window opens but is never filled out.
If I do the same thing via ssh as in
    ssh -X localhost Downloads/Cura/Ultimaker_Cura-4.4.1.AppImage
I get the same splash display followed by a startup window. This is very strange.

I have captured screenshots and several files created by Cura in the attached archive. How do I figure out what's wrong?
[cura.tar.gz](https://github.com/Ultimaker/CuraEngine/files/4112292/cura.tar.gz)

sk8nfool, 2020-01-25:
I goofed on the attachment. This should be better.
[cura.tar.gz](https://github.com/Ultimaker/CuraEngine/files/4112724/cura.tar.gz)

sk8nfool, 2020-01-25:
I compared the memory maps of each copy of cura: one running broken and the other running fine and noticed that they had different display drivers
> /usr/lib/x86_64-linux-gnu/dri/nouveau_dri.so (broken)
> /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so (working)

so I set DISPLAY on both to ":0" (which worked because I was the same user) and both ran with the nouveau_dri.so driver - and both hung. I also tried the reverse (started from local terminal with display set to remote display (eg "DISPLAY=localhost:12.0 Downloads/Cura/Ultimaker_Cura-4.4.1.AppImage")) and it too worked.

Ghostkeeper, 2020-01-27:
Could you try clearing your `~/.qtshadercache` folder?

sk8nfool, 2020-01-27:
On my system there is no ~/.qtshadercache but there is a
~/.cache/qrshadercache so instead I cleared that by removing any files in
the folder.
That fixed it. Cura started up as expected. Since I'd been running it over
a localhost ssh link Cura just started up normally (in a untitled empty
project). Si I did a complete restart by

   - flushing the QT cache as you suggested (but where I found it)
   - removing ~/.config/cura/4.4
   - removing ~/.local/share/cura/4.4

And cura started with its introductory startup sequence.
So I think you got it!

But now the question is why is the QT cache causing such bad things (like
hanging)? Is this a bug/problem with QT (presumably the version in the
AppImage) or the way Cura is using the QT libraries (or both).

At this point I don't have copies of the cache data 'cause I'm an idiot and
didn't protect the data.

Thanks again

Maybe Cura can tell QT to use a new cache? Or

Ghostkeeper, 2020-01-29:
Ah sorry yeah, I misremembered the location. Indeed for me it's also in `~/.cache/qtshadercache`.

Ghostkeeper, 2020-01-29:
> But now the question is why is the QT cache causing such bad things (like
hanging)? Is this a bug/problem with QT (presumably the version in the
AppImage) or the way Cura is using the QT libraries (or both).

Honestly we don't know why this happens. But it's not like we're instructing Qt to literally draw blanks once in a while, so the blame lies with either a bug in Qt or with a usability problem. I think we're not doing a whole lot of special things with regards to rendering and we've never touched Qt's shader caching system so I'm leaning towards Qt bug. However I haven't found any other bug reports about it yet, and it could be that we should have a `try:` `except` block wrapped around some things. Of course we are hidden behind the PyQt layer as well so we're not directly talking to Qt.

rburema, 2020-01-29:
> At this point I don't have copies of the cache data 'cause I'm an idiot and didn't protect the data.

Don't call yourself that just for not making a backup of something that most people don't even know exists, and _shouldn't_ give any problems anyway :-)

Anyway, there _might_ be a workaround for this in the future version 4.5. (Since shader cache corruption was a candidate to be the cause of an annoying windows hang, and although it turned out not to be, we still snuck in a fix, since it might be an issue in the future.) It's not on by default though, and not in the Cura interface. You should be able to manually set `force_empty_shader_cache = True` under the `[view]` heading in the preference(s) file when that version comes out.

(Devs: All I had to do for that was call `self.setAttribute(Qt.AA_DisableShaderDiskCache)` in QtApplication if that preference is set. There's a number of bug reports floating around about shader caches crashing stuff in other programs as well, so I thought it was a good candidate for the mystery bug.)