Ultimaker/CuraEngine #1189: circle bridges with inner ring being printed before outer ring and skin

fractalfederation, 2020-01-28:
[image]

Using latest Cura release.

It happens with circles or squares, any time it's a bridge wrapping around like a circle. The inner ring gets printed first, then the outer ring, then the skin infill. Might help to have outer, skin, then inner or just do no skin and do nothing but rings working from outer to inner.

smartavionics, 2020-01-28:
Cura doesn't attempt to solve this conundrum. You must either use support or, as I normally do, change the model to place a single layer of skin over that rectangular hole. I would then cut away the skin in the centre after the print has finished.

fractalfederation, 2020-01-28:
I didn't think about covering the hole with a thin layer then just cutting it out. Thanks! Any plans to handle this conundrum in the future, maybe bridge settings like Walls Only and Outer First or something?

smartavionics, 2020-01-28:
Or maybe a setting to automatically fill the skin where the hole is? I think that would produce better results.

Ghostkeeper, 2020-01-30:
That technique is called [sacrificial bridging](https://hackaday.com/2017/10/17/sacrificial-bridge-avoids-3d-printed-supports/). Useful trick to learn. Another alternative is to use staggered bridges like this: https://www.reddit.com/r/3Dprinting/comments/en7jk9/how_to_reduce_support_material_on_3d_prints_using/

We have an issue in our backlog to add a setting to print skin before the walls. It's a manual one though.

> maybe bridge settings like Walls Only and Outer First or something?

What do you mean with "Walls Only and Outer First"? Will it print only the walls? There is already a setting to print the outer wall first or not (the outer wall being the inner ring in your case).

fractalfederation, 2020-01-31:
> That technique is called [sacrificial bridging](https://hackaday.com/2017/10/17/sacrificial-bridge-avoids-3d-printed-supports/). Useful trick to learn. Another alternative is to use staggered [...]

i mean make the circular bridge nothing but rings, working from outside to the inside, so it already have a side to connect to. maybe linear infill versus zigag so its just a bunch of rings with smaller and smaller diameter

Ghostkeeper, 2020-02-04:
Hmm, I see. It'll need to convert a square shape to a round shape at some point though.