Ultimaker/CuraEngine #1191: [4.4.1] Strange pathways in the middle of wall of hollow cylinder on some layers

LightTemplar, 2020-01-29:
**Application Version**
4.4.1

**Platform**
KDE Neon, Linux

**Steps to Reproduce**
I added an STL with a hollow cylinder as part of it. Sliced and looked layers. 

**Actual Results**
Some layers had no circles in the middle of the round wall, but something different. 

**Expected results**
all pathways on all layers of the hollow cylinder are circular.

**Additional Information**
Wall line count is 4, so it is enough to make all cylinder from walls.
 
Project: https://we.tl/t-WOTjhi49Im
Original STL: https://we.tl/t-tkGnd3l2eb

For test purposes I made a simple cylinder. It has a slightly different pattern in the middle, but still has.
Its project: https://we.tl/t-kvHReyehnf
Simple Cylinder STL: https://we.tl/t-vAlkNDiFHX

[image]
[image]

LightTemplar, 2020-01-29:
For test purposes I made a simple cylinder. It has a slightly different pattern in the middle, but still has.
Its project: https://we.tl/t-kvHReyehnf
Simple Cylinder STL: https://we.tl/t-vAlkNDiFHX

smartavionics, 2020-01-29:
Unfortunately, the wall gap filling (and thin wall printing) in UM Cura does do what you have observed. It's a known problem. If you wish to try one of my Cura releases which can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0 you will find that they handle the gap filling with fewer artifacts. Please read the README.md file there for more information. Note that in my latest releases I added some extra options to the gap fill menu. Unfortunately, I goofed and they don't work as I expected but the normal "Everywhere" option does work OK. Here's what I get when I sliced your project.

[image]

LightTemplar, 2020-01-30:
Unfortunately, your release is not usable for me, as it has some strange font glitches. And I'm not sure, if it lies within the app, or within my system, because I saw the same problem also with teamviewer. Google didn't help much :(

[image]

smartavionics, 2020-01-30:
That's a shame, it must be because I am building my releases using QT 5.13.2 rather than 5.10 or whatever UM are currently using. I know they are looking at using a more recent version of QT so it could be that their future releases wouldn't work right for you either. Could you please run the AppImage from a terminal window and see if any interesting messages appear there. Thanks.

smartavionics, 2020-01-30:
Could you please open an issue at https://github.com/smartavionics/Cura so we can discuss this font/UI problem.

A quick search found me https://cullmann.io/posts/kde-qt-highdpi-scaling/ maybe it's relevant.

smartavionics, 2020-01-30:
Also, this could be relevant https://doc.qt.io/qt-5/highdpi.html

LightTemplar, 2020-01-30:
Sure thing! https://github.com/smartavionics/Cura/issues/41

Ghostkeeper, 2020-01-31:
Duplicate of https://github.com/Ultimaker/Cura/issues/5110. Since that issue was raised we've made some improvements but it's still a bug. This is an ongoing project for us. Last week we've had a POC implementation brought to us that would fix this issue in a good way. Now we need to make that release-ready. Please continue discussion in the other thread.

closed 2020-01-31 by Ghostkeeper

Ghostkeeper, 2020-02-03:
By the way from source I'm running Qt 5.13.2 as well, and I'm not having that visual glitch.