Ultimaker/CuraEngine #1193: [4.4.1] Regression test failure on arm-* and ppc64: InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L350__0, where GetParam() = InfillTestParameters_P8_Z0_C1_L350__0(input OK)

df7cb, 2020-01-30:
**Application Version**
4.4.1

**Platforms**
Debian Linux: arm64, armel, armhf, ppc64el

**Steps to Reproduce**
Building 4.4.1 and master as of e904e260716 on Debian fails on various architectures (only amd64, sparc64, and x32 are fine)

On arm64, armel, armhf, and ppc64el the problem is:
```
[       OK ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P7_Z1_C1_L1200__4 (1 ms)
[ RUN      ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C0_L350__4
[       OK ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C0_L350__4 (0 ms)
[ RUN      ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L350__4
/<<PKGBUILDDIR>>/tests/InfillTest.cpp:253: Failure
Expected: ((coord_t)out_infill_area) > ((coord_t)min_expected_infill_area), actual: 416261650 vs 776258802
Infill area should be greater than the minimum area expected to be covered.
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L350__4, where GetParam() = InfillTestParameters_P8_Z0_C1_L350__4(input OK) (0 ms)
[ RUN      ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C0_L350__4
[       OK ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C0_L350__4 (0 ms)
[ RUN      ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L350__4
/<<PKGBUILDDIR>>/tests/InfillTest.cpp:253: Failure
Expected: ((coord_t)out_infill_area) > ((coord_t)min_expected_infill_area), actual: 416261650 vs 776258802
Infill area should be greater than the minimum area expected to be covered.
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L350__4, where GetParam() = InfillTestParameters_P8_Z1_C1_L350__4(input OK) (0 ms)
[ RUN      ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C0_L400__4
[       OK ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C0_L400__4 (0 ms)
[ RUN      ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L400__4
/<<PKGBUILDDIR>>/tests/InfillTest.cpp:253: Failure
Expected: ((coord_t)out_infill_area) > ((coord_t)min_expected_infill_area), actual: 364345800 vs 676333491
Infill area should be greater than the minimum area expected to be covered.
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L400__4, where GetParam() = InfillTestParameters_P8_Z0_C1_L400__4(input OK) (0 ms)
[ RUN      ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C0_L400__4
[       OK ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C0_L400__4 (0 ms)
[ RUN      ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L400__4
/<<PKGBUILDDIR>>/tests/InfillTest.cpp:253: Failure
Expected: ((coord_t)out_infill_area) > ((coord_t)min_expected_infill_area), actual: 364345800 vs 676333491
Infill area should be greater than the minimum area expected to be covered.
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L400__4, where GetParam() = InfillTestParameters_P8_Z1_C1_L400__4(input OK) (1 ms)
[ RUN      ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C0_L600__4
[       OK ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C0_L600__4 (0 ms)
[...]
[----------] Global test environment tear-down
[==========] 1080 tests from 1 test suite ran. (2799 ms total)
[  PASSED  ] 1058 tests.
[  FAILED  ] 22 tests, listed below:
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L350__0, where GetParam() = InfillTestParameters_P8_Z0_C1_L350__0(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L350__0, where GetParam() = InfillTestParameters_P8_Z1_C1_L350__0(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L350__1, where GetParam() = InfillTestParameters_P8_Z0_C1_L350__1(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L350__1, where GetParam() = InfillTestParameters_P8_Z1_C1_L350__1(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L400__1, where GetParam() = InfillTestParameters_P8_Z0_C1_L400__1(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L400__1, where GetParam() = InfillTestParameters_P8_Z1_C1_L400__1(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L350__2, where GetParam() = InfillTestParameters_P8_Z0_C1_L350__2(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L350__2, where GetParam() = InfillTestParameters_P8_Z1_C1_L350__2(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L400__2, where GetParam() = InfillTestParameters_P8_Z0_C1_L400__2(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L400__2, where GetParam() = InfillTestParameters_P8_Z1_C1_L400__2(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L350__3, where GetParam() = InfillTestParameters_P8_Z0_C1_L350__3(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L350__3, where GetParam() = InfillTestParameters_P8_Z1_C1_L350__3(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L400__3, where GetParam() = InfillTestParameters_P8_Z0_C1_L400__3(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L400__3, where GetParam() = InfillTestParameters_P8_Z1_C1_L400__3(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L350__4, where GetParam() = InfillTestParameters_P8_Z0_C1_L350__4(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L350__4, where GetParam() = InfillTestParameters_P8_Z1_C1_L350__4(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L400__4, where GetParam() = InfillTestParameters_P8_Z0_C1_L400__4(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L400__4, where GetParam() = InfillTestParameters_P8_Z1_C1_L400__4(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L350__5, where GetParam() = InfillTestParameters_P8_Z0_C1_L350__5(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L350__5, where GetParam() = InfillTestParameters_P8_Z1_C1_L350__5(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z0_C1_L400__5, where GetParam() = InfillTestParameters_P8_Z0_C1_L400__5(input OK)
[  FAILED  ] InfillTestcases/InfillTest.TestInfillSanity/InfillTestParameters_P8_Z1_C1_L400__5, where GetParam() = InfillTestParameters_P8_Z1_C1_L400__5(input OK)

22 FAILED TESTS

90% tests passed, 2 tests failed out of 20

Total Test time (real) =   4.33 sec

The following tests FAILED:
	  3 - InfillTest (Failed)
	 17 - PolygonUtilsTest (Failed)
Errors while running CTest
make[1]: *** [Makefile:86: test] Error 8
```

Full build logs are linked here: https://buildd.debian.org/status/logs.php?pkg=cura-engine&ver=1%3A4.4.1-1

onitake, 2020-01-30:
I'd wager a guess that quite a few people are running Cura on a Raspberry Pi or similar, so failures on arm and arm64 are serious enough to warrant an urgent fix. But failures on other archs should be addressed as well.

I'll try to trace the issue in a VM, but if you have any hints why the unit tests might fail, that would help a lot.

rburema, 2020-01-31:
It might be useful to uncomment line 9 (`//#define TEST_INFILL_SVG_OUTPUT`) in that particular test-file (InfillTest.cpp). It'll put some visual output in your tmp folder that might be useful during debugging the tests.

onitake, 2020-01-31:
Thanks, @rburema .

I can reproduce those failures, and here's the output:
[InfillTest.zip](https://github.com/Ultimaker/CuraEngine/files/4141083/InfillTest.zip)
The arm64 results have some very strange crooks that seem to dislodge the positioning of the lines and interrupt continuous patterns, which causes the unit tests to fail.
One example (InfillTestParameters_P8_Z0_C1_L350__4) as an image:
amd64:
[image]
arm64:
[image]

On the other hand, the non-failing tests look fine to me.

onitake, 2020-02-03:
Oh, and if you want to test-build on non-x86 architectures yourself, that's actually very easy:
Install `virt-manager` and `qemu-system-<arch>`, then grab an installation ISO from the Debian archive. Then just create a new VM in virt-manager and select the architecture you want.
arm and arm64 emulations work very well and even have decent performance.

Ghostkeeper, 2020-02-03:
Sorry, but we don't have the manpower to target anything other than x86-64 via GCC.

If you're interested in working on a fix, try looking into the precision of the built-in types. We should use `coord_t` for all coordinates which is fixed to `int64_t`, but here and there we might still be using `int` or `float`.

And it could be that the actual issue lies in [clipper](http://www.angusj.com/delphi/clipper.php) rather than CuraEngine. Clipper is no longer being maintained much, though it has a lot of forks where people still make bug fixes sometimes.

df7cb, 2020-02-03:
It's not a "simple" 32-bit problem because arm64 and ppc64el have the same problem. It's not endianness either. Also, all these platforms have proper IEEE754 floats, so that's unlikely a reason.

Which clipper fork would you recommend to be used?

df7cb, 2020-02-03:
Also, is there something we should do to make the build/test log more detailed, like cat'ing some files on error?

Ghostkeeper, 2020-02-04:
> Which clipper fork would you recommend to be used?

The embedded copy we have is version 6.4 of the original version from Angusj, with some modifications to make it compile with GCC9. The changes made for GCC9 include that line you indicate in that diff, as well as some documentation with strange characters. So that makes sense. If you're using GCC8 or lower, I think that your separate package should work fine. If you have a higher version of GCC, I recommend applying that [patch](https://github.com/Ultimaker/CuraEngine/commit/46d73aded5cb4e6107a66f410f0397b62c6ee257).

onitake, 2020-02-04:
From my C++ knowlege, it looks like the patch only makes the implicit default assignment operator explicit. It doesn't change any code and is probably only there to silence a nitpick compiler...

onitake, 2020-02-04:
I'll try to trace the problem, but my understanding of CuraEngine's algorithms is very limited. It may take a while.

onitake, 2020-02-06:
I haven't found the bug yet, but here's a little status report:
By tracing the program flow, I think I narrowed the problem down to `PolygonConnector::addPolygonSegment`. For the concentric circle case, it seems like it tries to join successive circles, but is cut short while tracing the second segment. This results in the circle being skipped, which can be clearly seen in the images. The tiny nooks are basically the segment joiners being connected together without the segment.

The longer I look at the problem, it seems like there's a logic error somewhere, not a math error. But for some reason, this logic error only manifests on certain architectures. Hmm.

df7cb, 2020-02-07:
https://github.com/Ultimaker/CuraEngine/blob/ad16c1bc6c6061836286598952e6fe38aa11669d/src/utils/PolygonConnector.cpp#L105-L114

`char` is unsigned on arm and powerpc, and hence returning -1 doesn't work. Making that function `int` (and `char dir` on line 84) should fix the problem.

https://wiki.debian.org/ArchitectureSpecificsMemo

rburema, 2020-02-07:
Thanks for the investigation all! :-)

closed 2020-02-07 by rburema

onitake, 2020-02-07:
@rburema I think a `signed char` or int8_t would have sufficed, as the returned value will only be -1 or 1.

commit 4d049c47 (2020-02-07): Be more compliant with non-x86 architectures.

Replace integer-like uses of char with int16_t, as char can be unsigned by default on some architectures (which means -1 cant be used as a signaling value).
Should fix #1193 at least.