Ultimaker/CuraEngine #1194: [4.4.1] Regression test failure on mips64el: GCodeExportTest.insertWipeScriptRetractionEnable

df7cb, 2020-01-30:
**Application Version**
4.4.1

**Platform**
mips64el Debian Linux

**Steps to Reproduce**
Building 4.4.1 on Debian unstable fails on various architectures (only amd64, sparc64, and x32 are fine)

On mips64el the problem is:
```
[       OK ] GCodeExportTest.insertWipeScriptOptionalDelay (0 ms)
[ RUN      ] GCodeExportTest.insertWipeScriptRetractionEnable
/<<PKGBUILDDIR>>/tests/GCodeExportTest.cpp:646: Failure
Expected equality of these values:
  std::string("G1 F120 E99")
    Which is: "G1 F120 E99"
  token
    Which is: "G0 F600 X2 Y1"
Wipe script should perform retraction with provided speed and retraction distance.
/<<PKGBUILDDIR>>/tests/GCodeExportTest.cpp:651: Failure
Expected equality of these values:
  std::string("G1 F180 E104")
    Which is: "G1 F180 E104"
  token
    Which is: ""
Wipe script should make unretraction with provided speed and extra prime volume.
/<<PKGBUILDDIR>>/tests/GCodeExportTest.cpp:653: Failure
Expected equality of these values:
  std::string(";WIPE_SCRIPT_END")
    Which is: ";WIPE_SCRIPT_END"
  token
    Which is: ""
Wipe script should always end with tag.
[  FAILED  ] GCodeExportTest.insertWipeScriptRetractionEnable (2 ms)
[ RUN      ] GCodeExportTest.insertWipeScriptHopEnable
[       OK ] GCodeExportTest.insertWipeScriptHopEnable (0 ms)
[----------] 29 tests from GCodeExportTest (10 ms total)

[----------] 4 tests from GriffinHeaderTestInstantiation/GriffinHeaderTest
[ RUN      ] GriffinHeaderTestInstantiation/GriffinHeaderTest.HeaderGriffinFormat/0
[       OK ] GriffinHeaderTestInstantiation/GriffinHeaderTest.HeaderGriffinFormat/0 (1 ms)
[ RUN      ] GriffinHeaderTestInstantiation/GriffinHeaderTest.HeaderGriffinFormat/1
[       OK ] GriffinHeaderTestInstantiation/GriffinHeaderTest.HeaderGriffinFormat/1 (0 ms)
[ RUN      ] GriffinHeaderTestInstantiation/GriffinHeaderTest.HeaderGriffinFormat/2
[       OK ] GriffinHeaderTestInstantiation/GriffinHeaderTest.HeaderGriffinFormat/2 (1 ms)
[ RUN      ] GriffinHeaderTestInstantiation/GriffinHeaderTest.HeaderGriffinFormat/3
[       OK ] GriffinHeaderTestInstantiation/GriffinHeaderTest.HeaderGriffinFormat/3 (0 ms)
[----------] 4 tests from GriffinHeaderTestInstantiation/GriffinHeaderTest (2 ms total)

[----------] Global test environment tear-down
[==========] 33 tests from 2 test suites ran. (13 ms total)
[  PASSED  ] 32 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] GCodeExportTest.insertWipeScriptRetractionEnable

 1 FAILED TEST

      Start  5: TimeEstimateCalculatorTest
 2/20 Test  #4: MergeInfillLinesTest .............   Passed    0.08 sec
      Start  6: SlicePhaseTest
 3/20 Test  #5: TimeEstimateCalculatorTest .......   Passed    0.09 sec
      Start  7: SettingsTest
 4/20 Test  #7: SettingsTest .....................   Passed    0.12 sec
      Start  8: ArcusCommunicationTest
 5/20 Test  #8: ArcusCommunicationTest ...........   Passed    0.07 sec
      Start  9: ArcusCommunicationPrivateTest
 6/20 Test  #9: ArcusCommunicationPrivateTest ....   Passed    0.09 sec
      Start 10: AABBTest
 7/20 Test #10: AABBTest .........................   Passed    0.03 sec
      Start 11: AABB3DTest
 8/20 Test #11: AABB3DTest .......................   Passed    0.03 sec
      Start 12: IntPointTest
 9/20 Test #12: IntPointTest .....................   Passed    0.02 sec
      Start 13: LinearAlg2DTest
10/20 Test #13: LinearAlg2DTest ..................   Passed    0.05 sec
      Start 14: MinimumSpanningTreeTest
11/20 Test #14: MinimumSpanningTreeTest ..........   Passed    0.03 sec
      Start 15: PolygonConnectorTest
12/20 Test #15: PolygonConnectorTest .............   Passed    0.14 sec
      Start 16: PolygonTest
13/20 Test #16: PolygonTest ......................   Passed    0.03 sec
      Start 17: PolygonUtilsTest
14/20 Test #17: PolygonUtilsTest .................   Passed    0.05 sec
      Start 18: SparseGridTest
15/20 Test #18: SparseGridTest ...................   Passed    0.04 sec
      Start 19: StringTest
16/20 Test #19: StringTest .......................   Passed    0.03 sec
      Start 20: UnionFindTest
17/20 Test #20: UnionFindTest ....................   Passed    0.02 sec
18/20 Test  #6: SlicePhaseTest ...................   Passed    2.16 sec
19/20 Test  #1: BuildTests .......................   Passed   12.63 sec
20/20 Test  #3: InfillTest .......................   Passed   95.72 sec

95% tests passed, 1 tests failed out of 20

Total Test time (real) =  95.75 sec

The following tests FAILED:
	  2 - GCodeExportTest (Failed)
Errors while running CTest
make[1]: *** [Makefile:86: test] Error 8
```

Full build log https://buildd.debian.org/status/fetch.php?pkg=cura-engine&arch=mips64el&ver=1%3A4.4.1-1&stamp=1580257421&raw=0
ia64 has the same problem: https://buildd.debian.org/status/fetch.php?pkg=cura-engine&arch=ia64&ver=1%3A4.4.1-1&stamp=1580295308&raw=0 (but unlike mips64el, ia64 is not a release architecture)

Other architectures are listed here: https://buildd.debian.org/status/logs.php?pkg=cura-engine&ver=1%3A4.4.1-1

Ghostkeeper, 2020-02-03:
I've been getting these errors myself for a while, too. They don't happen on our CI server, but only for me locally on several different computers (all are x86-64 though).

I don't think it's related to the instruction set or processor architecture, but something fishy is going on for sure.

df7cb, 2020-02-10:
With #1193 fixed, https://buildd.debian.org/status/logs.php?pkg=cura-engine&ver=1%3A4.4.1-2 looks much better now. We are left with:
* this one: #1194 affecting mips64el, ia64, and hppa. (Only mips64el is a release architecture)
* #1192 affecting i386 (possibly harmless?) rounding issues

onitake, 2020-02-18:
By the way: Docker supports running containers built for different architectures, as long the host can run binaries for the desired CPU architecture. Qemu can give you emulation for foreign architectures.
Sadly, docker doesn't distinguish image names, so you need to remove any with the same name first. You also need to enable experimental mode in dockerd.

Try:
```sh
sudo cat >/etc/docker/daemon.json <<EOF
{"experimental":true}
EOF
sudo systemctl restart docker
sudo apt install qemu-system-arm
docker rmi debian:sid
docker run -ti --rm --platform linux/aarch64 debian:sid
```
This should give you a nice ARM64 container for debugging/testing.

df7cb, 2020-03-14:
https://github.com/Ultimaker/CuraEngine/blob/dbf3d79a112894fdf9f3b9460923f85256f04b44/tests/GCodeExportTest.cpp#L646-L648

To debug this, I was trying to see if the three untested intermediate strings would bring any insight by comparing them on amd64 and mips64el, but my test build on the Debian mips64el porterbox didn't fail. (While the official buildd still fails on 4.5.0.)

For reference, the three strings in these lines are:
* "G0 F600 X2 Y1"
* "G0 X2.5 Y1"
* "G0 X1 Y1"

Ghostkeeper, 2020-08-14:
I found a place where it read from uninitialised memory in the test, which caused it to fail in the same way for me. I fixed that, so hopefully it'll also be fixed for you.

closed 2020-08-14 by Ghostkeeper

df7cb, 2020-08-14:
Thanks! Will give it a try.

df7cb, 2020-08-19:
mips64el is happy now. Thanks!

https://buildd.debian.org/status/package.php?p=cura-engine

commit 3744d024 (2020-08-14): Fix test reading from uninitialised memory

On some systems (or maybe the debug build type?) the max retraction count got initialised to 0, which made the test fail because it skipped the expected retraction.

Fixes #1194.