Ultimaker/CuraEngine #1196: [CuraEngine 4.4.1] - Support Interface not generated when Support Interface Extruder not T0 or T1

mkudzia84, 2020-01-31:
When selecting Support Interface Extruder, support interface is not generated unless Extruder 1 or Extruder 2 are selected

CuraEngine 4.4.1
Setup 4 extruder printer.

Ghostkeeper, 2020-02-04:
I'm unable to reproduce this problem. These are my steps:
* Add a Custom FFF Printer with 4 extruders.
* Load a different coloured material in each extruder so that you can see the difference in Layer View.
* Load a model that has overhang.
* Enable Support.
* Set the Support Interface Extruder to the 3rd extruder.
* Enable Support Interface.

The result is that the support interface is printed with the 3rd extruder:
[image]

Are you sure that support interface is enabled at all? It's crucial that you enable the support interface after switching the support interface extruder, because the enabled/disabled setting can be different per extruder, depending on the profile (e.g. PVA will enable it by default).

mkudzia84, 2020-02-04:
Ok,

I played around with the GUI and it seems it's not a CuraEngine issue but the GUI issue;
I think I found the root cause;
Even when choosing the different extruder tabs, the Support drop down only shows information related to the Extruder selected as the top level "Support Extruder"

I.e. you switch to tab for Extruder 3:
And the settings are:
- Support Extruder : Extruder 2
- Interface Extruder : Extruder 3
All of the following settings in that tab are shown for extruder 2 (top level Support extruder)
There is no clear breakdown here on which settings are 'extruder specific' - i.e. interface layer settings
and which settings are global (I do have to say PrusaSlicer organises settings much better here...)

So in order to enable the interface layer for specific extruder
a) selecting top level tool tabs has no effect
1) select appropritate extruder as top level "Support Extruder" in support tab
2) set appropriate settings
3) Select back the desired top level "support extruder"
4) Select the desired interface extruder

Clunky and not self documenting to say the least...

Ghostkeeper, 2020-02-06:
> There is no clear breakdown here on which settings are 'extruder specific' - i.e. interface layer settings
and which settings are global

The distinction is that the global settings have this chain link icon on them.

Settings that apply to the support interface will take their values from the support interface extruder (with the appropriate defaults for the material and nozzle loaded in there). That means that if you switch the support interface extruder, the settings can change. As a result, if you want to change the support interface settings, you should select the support interface you want to use first, or else it's going to snap back to the defaults for the extruder you're changing to.

So changing the support extruder to the support interface extruder, then changing settings, then changing the support extruder back and the support interface extruder forth again is unnecessary. Just selecting the support interface extruder first is enough.