Ultimaker/CuraEngine #1199: Problem in Cura running source from Windows-10

Vishnu-Chittan, 2020-02-10:
Hello All, 
I am trying to configure Cura build sources on my Windows 10 Hp laptop. I followed all the instructions given in the GH  wikipage. but some how i am not able to get it done. can you please help me in this regard. 

**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

[image]

rburema, 2020-02-11:
Are you running this from the 'VS2015 x64 Native Tools Command Prompt' or 'VS2019 x64 Native Tools Command Prompt' (as opposed to plain cmd or PowerShell)?

Vishnu-Chittan, 2020-02-11:
> Are you running this from the 'VS2015 x64 Native Tools Command Prompt' or 'VS2019 x64 Native Tools Command Prompt' (as opposed to plain cmd or PowerShell)?
@rburema 
I am using direct CMake GUI.

Vishnu-Chittan, 2020-02-11:
@rburema ...
Though I tried in native command window got the same error.
Dont know the reason for failure ...

[image]

rburema, 2020-02-11:
You should start from either of those native tools command prompts. (You can probably start CMake gui as well with `cmake-gui ..`).