Ultimaker/CuraEngine #1203: [4.5 BETA] 'Minimum Layer Time' inconsistent

Timocop, 2020-02-15:
**Application Version**
4.5 BETA

**Platform**
OS: Windows 10 x64 1909 (Build 18363.592)
AV: Windows Defender

**Display Driver**
Nvidia GeForce RTX 2060 Super (v441.08)

**Steps to Reproduce**
 - Open a model with small parts, like the benchy (default preset).
 - Set the "minimum time" in cooling to a high number.
 - (Optional) Set 'Overhanging Wall Speed' or 'Small Feature Speed' to 50% or something.
 - See the mess of speeds variations.

**Actual Results**
 - When setting 'Minimum Layer Time' to a giant value only some parts print at 'Minimum Speed'.

**Expected results**
- When setting 'Minimum Layer Time' to a giant value all parts should print at 'Minimum Speed'.

**Additional Information**
Cura 4.5 BETA:
[image]
[image]
Cura 3.6 (Same issue)
[image]
Cura 3.4.1
[image]

Ghostkeeper, 2020-02-20:
Indeed, the entire layer is slowed down as a whole until one of the speeds hits the Minimum Speed setting. So for instance if your Outer Wall Speed is 30mm/s and the Inner Wall Speed is 40mm/s So for instance if you set the Outer Wall Speed to be equal to your Minimum Speed, then the Minimum Layer Time setting won't have any effect at all any more since it can't slow down the outer walls.

You're asking here that the lines are individually slowed down to the Minimum Speed. If some lines are printed faster than others, then at least the fast lines can still be slowed down until the Minimum Layer Time is reached. Indeed that makes sense from a print quality perspective, I think.

Timocop, 2020-02-20:
But why would it be the *slowest* speed and not the *fastest* speed that hits the Minimum Speed setting. Wouldn't it make more sense to search the fastest line in the layer and use that speed to scale everything down instead and cap it by the Minimum Speed if it falls below? The whole point of this feature is to slow down fast lines. So why would you use the slowest line in the layer?
Its like the police slowing down cars on a highway that barely go above the speed limit but ignore the fastest cars on the highway.

Especially with all the new features Cura got over the years that change lines speeds. The Minimum Speed logic seem quite outdated and the definition quite confusing. I always thought the logic would be as simple as ``line_speed = max(scaled_line_speed, minimum_speed)``.

Example:
Lets say im printing steady at 60mm/s but for 1 path Cura created it prints at 10mm/s (For example due to Overhang Wall Speed or slow Top/Bottom). Everything will not be scaled down because that one path hit the minimum speed? That makes not much sense. Like in this example:
[image]
I already use 'Small Feature Speed' as a replacement instead which worked way better than 'Minimum Layer Time' because it behaves way to unpredictable.

*Sorry for the abuse of the english language*

Ghostkeeper, 2020-02-25:
> But why would it be the slowest speed and not the fastest speed that hits the Minimum Speed setting. Wouldn't it make more sense to search the fastest line in the layer and use that speed to scale [...]

The reason is that if you choose the fastest speed, then other lines would be lower than the Minimum Speed and that would degrade the material because it's moving at 0.1mm/s with a hot nozzle.

The problem is that it's slowing down the whole layer with the same factor, rather than choosing to slow down some lines individually. I suppose this was done to make it easier to predict how much you'd need to slow down the layer in order to reach the minimum layer time.

The logic right now, in pseudocode, is something like:

```
total_layer_time = sum([line.duration() for line in layer])
if total_layer_time < minimum_layer_time:
    speed_factor = total_layer_time / minimum_layer_time
    for line in layer:
        if line.speed * speed_factor < minimum_speed:
            speed_factor = minimum_speed / line.speed
    for line in layer:
        line.speed *= speed_factor
```

So if the minimum layer time is more medium (rather than 9999) you'd want to slow down the lines *just enough* so that it reaches exactly the minimum layer time. This is difficult if the lines are slowed down by different factors.

Timocop, 2020-02-25:
> The reason is that if you choose the fastest speed, then other lines would be lower than the Minimum Speed

Exactly this. Then everything thats *below* is capped at the minimum speed.

> it's moving at 0.1mm/s with a hot nozzle

Cap the speed with minimum speed after you applied the factor.

I think we should ignore the individual layer speed for now, it doesnt seem to be an issue realy.
But i have an issue how minimum line speeds are calculated.
What i think when i read 'Minimum Speed' is everything that falls bellow this minimum speed will be capped. E.g. if you slowdown a line so much that it prints at 0.1mm/s but you set minimum speed to 10mm/s. So it speeds all slow lines the are below it to 10mm/s. But instead, it does something completely different.

I think this pseudocode explains more than my english 😅 
Your modified pseudocode (I dont know python but should be clear):
```Python
total_layer_time = sum([line.duration() for line in layer])
if total_layer_time < minimum_layer_time:
    speed_factor = total_layer_time / minimum_layer_time
    for line in layer:
        # Slow down line speed by factor and make sure we never fall below minimum speed.
        line.speed =  math.max(line.speed * speed_factor, minimum_speed)
```

Ghostkeeper, 2020-02-27:
I understand that approach, but it's still problematic, because the minimum layer time will then no longer be applied. After all, you're increasing the speed of some of the lines, so the total time of the layer will be too short. The minimum layer time can still be reached then by further reducing the speed of other lines. This is where things get complex, because it would sort of require iterative re-calculation of the print time.

Ghostkeeper, 2020-03-02:
Developers, see issue CURA-7271.