Ultimaker/CuraEngine #1204: Missing default_value in recent configs?

richfelker, 2020-02-16:
I use CuraEngine standalone without Cura GUI. I tried upgrading to git master a few days ago, and usually pull updated json def files from Cura too, since otherwise there are often new settings without a value set. I was surprised to find this time that I got a lot of warnings about options lacking a default_value. What is the intended usage pattern for dealing with this? Is it something that's expected to be fixed on the Cura side, or do command line users need a workflow to generate flattened configs with default_value settings?

Ghostkeeper, 2020-02-25:
As far as I know, [every setting](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json) has a `default_value` property. Both the front-end and the back-end check this.

Which JSON file were you loading in? What was the command you executed on the command line?

richfelker, 2020-02-25:
OK, I figured out where this came from. The errors I'm seeing are all for settings from the printer profile not the base `fdmprinter.def.json`, which indeed has all the `default_value`s. `creality_ender3.def.json` was refactored to use `creality_base.def.json`, and the latter completely lacks `default_value`s. Maybe it should be fixed, but for now I'll just stop using it and copy the few old settings I actually wanted out of the old one to my own file.

Ghostkeeper, 2020-02-28:
The settings in `creality_base.def.json` are only overrides though. `creality_base` inherits from `fdmprinter` which provides the `default_value`s, so I would expect that it still finds the correct defaults...

richfelker, 2020-02-28:
I would expect that too, but for some reason it didn't seem to be happening. If you can't reproduce it I'll write up a detailed report on how to make it happen. (And in any case, the profile isn't working right with standalone CuraEngine when it only has `value` not `default_value` for the settings. The old Ender 3 profile had `default_value` for most and was only lacking it for a few things.)

Ghostkeeper, 2020-03-03:
CuraEngine's command line interface doesn't read the `value` properties since that is a Python expression. CuraEngine doesn't have a Python interpreter.

We develop the command line interface for CuraEngine for use as debug mode to verify the connection between the front-end and the back-end, and to make it easier to run through CuraEngine with a debugger. For that purpose we don't need the `value` properties. If another application wants to use CuraEngine as slicing engine that's fine according to the license but they'd need to provide `-s` arguments or JSON files that have the `default_value` working correctly.

Ghostkeeper, 2020-03-03:
I've tried this command:

    /home/ruben/Projects/CuraEngine/build/CuraEngine slice -j creality_base.def.json -l /home/ruben/Projects/SlicerTestModels/basic/cube.stl

It seems to output the correct g-code without complaining about `default_value`s missing. So indeed I can't reproduce the issue at the moment.