Ultimaker/CuraEngine #1207: Tree supports placement: Touching Build plate ommits support points above model

seiomuu, 2020-02-20:
**Application Version**
4.4

**Platform**
Windows

**Steps to Reproduce**
Any model where there's support needed above another piece of the model

**Actual Results**
Even when it could reach areas when touching build plate it ignores many support points because it'd prefer to base them on top of previous model parts

**Expected results**
It would route the tree support point to the build plate when possible

mahtDFR, 2020-02-26:
[image]

Interesting idea, but this presents issues. For a very complex geometry this can create a path planning problem which can't be solvable in general. The diagram above shows some examples.

seiomuu, 2020-02-26:
If it can't find a base without going upwards it'll just not create support there. When Cura generates support it makes 2 categories, and it always takes the easiest path there and puts it into that category. When you then selects only one category, it'll omit everything from the other category without even trying to make it work by touching the build plate.

It's basically your top example, because it prefers to make the base ontop of the model, it won't try to generate a path from the build plate, even though it's clearly able to. So if you disable touching build plate you won't get any supports on the 2nd overhang.

Ghostkeeper, 2020-02-27:
I think I understand what you mean, but I'm not able to reproduce the issue. Here is a model with your case where Support Placement is set to Everywhere:
[image]

Here is the same model with Support Placement set to Buildplate Only:
[image]

So for Buildplate Only, it routes the branches that can reach the build plate to the build plate, and the rest are omitted. That seems like your expected behaviour, right?

Is there a specific model that you found this issue with? Perhaps a project file would help to debug it.