Ultimaker/CuraEngine #1208: Feature Request: Tree Infill

Daniel-Khodabakhsh, 2020-02-22:
Requesting infill type where tree supports are printed inside the body of the model such that the tips of the branches internally intersect portions of the body which exceed a specified overhanging angle.

This would optimise the amount of material used because branches can come from the wall sides instead of the floor.

Settings can include the same kinds of settings for tree supports.

The image below illustrates what I'm proposing. The red lines indicate the outer shell of the object, and the orange lines is the infill.

[image]

mahtDFR, 2020-03-03:
Thanks for the suggestion. After some discussion we decided this is not a development priority for us right now, so it's deferred.

Ghostkeeper, 2020-03-03:
The best alternative right now is to use Gradual Infill. It works very similarly to this suggestion and is more generic to work with each pattern.

Daniel-Khodabakhsh, 2020-03-03:
Thanks for considering my suggestion!
Would I be able to submit a pull request for this? May take me a while to get ramped up on the code base but I can give it a try.

Ghostkeeper, 2020-03-09:
Yeah. You'd need to look mainly into the code base for CuraEngine, but then also add a setting to `resources/fdmprinter.def.json` in the Cura repository so that it can be activated.

Ghostkeeper, 2020-03-09:
Actually, it's probably better to allow Infill Support to work properly with Tree Support.

Daniel-Khodabakhsh, 2021-12-04:
Thanks for adding lightning infill. Closing this ticket.

closed 2021-12-04 by Daniel-Khodabakhsh