Ultimaker/CuraEngine #1209: curaengine use

abtionlove, 2020-03-03:
When curaengine works from the command line, how can the parameters set after - s work? That is, how to use - s on the command line?

Ghostkeeper, 2020-03-09:
From the documentation:

```
  -s <setting>=<value>
	Set a setting to a value for the last supplied object, 
	extruder train, or general settings.
```

So basically like `-s layer_height="0.1"`.

closed 2020-03-09 by Ghostkeeper

abtionlove, 2020-03-09:
For example, to set the layer height, the JSON file has already set the layer height. Now use - s to set the layer height once, without modifying the JSON file, so the final layer height is the value in the JSON file, or the value after - S? How can I make the value after - s work? Thank you.

Ghostkeeper, 2020-03-12:
From my understanding, this is stateful. If you say:

    -s layer_height="0.1" -j ultimaker3.json

then the layer height in ultimaker3.json would override the 0.1 you provided. But if you say:

    -j ultimaker3.json -s layer_height="0.1"

then the customised layer height wins.