Ultimaker/CuraEngine #1213: Inconsistent number of layers calculation

alexjx, 2020-03-07:
**Application Version**
4.5

**Platform**
Windows

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

0. With the following settings:
    initial layer height of 0.27
    layer height of 0.2
1. Slice two cubic objects with 20mm and 21mm height.
2. Slice one cubic object with 20mm height.

**Actual Results**
When slicing alone, the 20mm cubic ends with 99 layers, the one sliced with another higher object ends 100 layers.
[image]
[image]

**Expected results**
The layers of the two cubics should have the same number of layers.

**Additional Information**
It seems that the calculation of the max number of layers doesn't consider the rounding correctly.
https://github.com/Ultimaker/CuraEngine/blob/3aa04cf85ec118945aebde3bdba00efd52a453a0/src/FffPolygonGenerator.cpp#L135
Which should be 
```
slice_layer_count = (storage.model_max.z - initial_layer_thickness + layer_thickness - 1) / layer_thickness + 1;
```

Is this intended or a bug?

Ghostkeeper, 2020-03-10:
Looks like a bug to me. Aside from Adaptive Layers, the number of layers in one object should not be influenced by the presence of other objects in the scene.

It doesn't reproduce with a 1cm cube, but indeed it does reproduce with 2cm cubes. The correct calculation should be:

* 1 layer with initial layer height (0.27mm):
* 19.73mm left for the rest of the layers. It's slicing the first of these at a relative height of 0.1mm (half a layer height). Absolute height is 0.37mm then, taking the initial layer into account.
* The remaining 19.63mm divides by 0.2mm layer thicknesses 98.15 times, rounded down to 98 times. The remaining 0.15 layers is above the centre-plane of the layer. So as a result the final model is 0.35 layers too thin.
* Together with the initial layer and the first normal-thickness layer that ends up at 100 layers.

In your calculation it would end up at `(20 - 0.27 + 0.2 - 1) / 0.2 + 1 =` 95 layers so we are disagreeing on the formula here.

alexjx, 2020-03-11:
Hmm...  I could reproduce this with 10mm and 20 mm cubic.

The formula should be correct if consider the values a not in mm but micron of integer form.
The idea is to take the ceiling of `height / layer thickness`.
Therefore with 20mm cubic, it should be as `(20000 - 270 + 200 - 1) / 200 + 1 = 100` (fractional part discarded).

mahtDFR, 2020-03-12:
This is on the backlog as CURA-7296

nallath, 2021-11-08:
We recently merged some fixes regarding this. I was unable to reproduce this when using master.

closed 2021-11-08 by nallath