Ultimaker/CuraEngine #1222: Does CuraEngine can be invoked on Android Studio?

Yzhao1125, 2020-03-24:
**Application Version**

CuraEngine 4.2.0

**Display Driver**

**Steps to Reproduce**

I build CuraEngine on AS with its own CMakelist as .a library by linked a .so library. Then, in my project, the JNI interfaces is written as follow:
'''extern "C"
JNIEXPORT jint JNICALL
Java_com_example_CureEnginelib_CuraEnginelib_CuraEngine(JNIEnv *env, jclass clazz, jstring string) {
    // TODO: implement CuraEngine()
    char *myargv[13]={"a","slice","-v","-j","jsonfile/fdmextruder.def.json","-o","output/test.gcode","-e1","-l","stl/cube.stl"};
    int myargc = 13;
    using namespace cura;
    cura::Application application;
    application.run(myargc,myargv);
    return myargc;
}'''

**Expected results**

I would like to invoke CuraEngine on Android Studio project. Is it able to realize? 

**Additional Information**

I run it on an Android AVD with armeabi v7a, but the following is mentioned:
'''SIGSEGV (signal SIGSEGV: invalid address (fault address: 0xfffffff4))''' 
and at GCodeExport::GcodeExport(), it is always mentioned that the following file does not belong to any project target,code insight features might not work properly.
...Android\Sdk\ndk\20.0.5594570\toolchains\llvm\prebuilt\windows-x86_64\sysroot\usr\include\c++\v1\ostream

rburema, 2020-03-24:
I _think_ we've only ever build CuraEngine as a standalone executable.

I'm a bit confused though, does this happen when you run or when you try to compile it?

Also, doesn't myarg[c|v] only contain 10 parameters here, not 13?

Yzhao1125, 2020-03-25:
> Also, doesn't myarg[c|v] only contain 10 parameters here, not 13?  
Sorry, I give a wrong myargc. The correct one is:
char *myargv[13]={"CuraEngine","slice","-v","-j","jsonfile/fdmextruder.def.json","-v","-j","jsonfile/fdmprinter.def.json","-o","output/test.gcode","-e1","-l","stl/cube.stl"};
But the above problems still exist. 
I compile CuraEngine as a static library using cmakelist without executable. Instead, link the static library with a dynamic library.

Ghostkeeper, 2020-03-26:
So our CMake first builds CuraEngine as a library and then links it to a Main function in order to execute. This was done because we can then link the same library to the automated tests. If you're using this library, you only need to look at the [main](https://github.com/Ultimaker/CuraEngine/blob/master/src/main.cpp) function. Basically, call `cura::Application::getInstance().run(argc, argv)`. You were missing the `getInstance()`?

Keep in mind though that we always compile CuraEngine for x86_64, not ARM. I don't know how it will run on your phone.

thevoiceless, 2020-04-12:
@Yzhao1125 If you get this figured out, can you share a sample application? I am also trying to integrate CuraEngine into an app