Ultimaker/CuraEngine #1223: Mutliple Stl Support from CommandLine

dombmann, 2020-03-24:
**Application Version**
4.4

**Platform**
Windows 10 

**Steps to Reproduce**
`CuraEngine.exe slice -v -j somedef.def.json -e0 -l stl1 -l stl2 -o mygcode.gcode`

**Actual Results**
stl1 and stl2 are overlapping 

**Expected results**
stls should not overlap

**Additional Information**
This is more of a general question if the curaEngine supports multiple stl in a slice command (from cmd)  and if so, how to avoid that the sliced stls overlapp 

Resulting GCode in Cura: 
[image]

rburema, 2020-03-24:
This happens because both stl files contain the origin. You can work around this by editing the stl file to have an offset from center. I'll ask around to see if this is a new feature or if something like this is already supported.

rburema, 2020-03-25:
Turns out this is already possible: add `-s mesh_position_y=150` (for example) to the command line. There's also a rotation matrix should you need it.

[This thread](https://github.com/Ultimaker/CuraEngine/issues/220) should explain it. (Be sure to read it all the way to the end.)

This makes it a duplicate, and I'll close it as such (any discussion should continue in the previously mentioned thread).

closed 2020-03-25 by rburema