Ultimaker/CuraEngine #1229: Arcus with Compiling Cura Engine

ArthurKretzer, 2020-04-01:
Hello there!

I'm working with Windows 10 and CMake. I have succeeded to build Protobuf 3.11.4 (following this tutorial https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Windows) and libArcus with -DBUILD_PYTHON=OFF (https://github.com/Ultimaker/libArcus).

When I try to "cmake .." CuraEngine, the following occurs:

> Found Protobuf: D:/dev/protobuf-3.11.4/cmake/build/release/libprotobuf.lib (found suitable version "3.11.4", minimum required is "3.0.0") 
> CMake Error at CMakeLists.txt:23 (find_package):
>   Could not find a configuration file for package "Arcus" that is compatible
>   with requested version "".
> 
>   The following configuration files were considered but not accepted:
> 
>     D:/dev/libArcus-master/build/ArcusConfig.cmake, version: 1.1.0 (32bit)
> 
> Configuring incomplete, errors occurred!

Why? I have installed libArcus lattest version.

rburema, 2020-04-01:
What's `Arcus_DIR` set to in CMake? 

P.S. Since you're using `-DBUILD_PYTHON=OFF`: If you're going to use C++ to use your slicing software, you could also find a way to build without libArcus, since that's the part responsible for communicating between the engine and the (python) front-end. (I assume your development is in GPL, since the engine is GPL, not LGPL, and the licence therefore would 'infect' your software if used as a library.)

ArthurKretzer, 2020-04-01:
I have included "D:\dev\libArcus-master\build" to system PATH.

Thanks for your P.S. since I need to use CuraEngine with Python so I will try to recompile it. And yes, my development is in GPL.

rburema, 2020-04-01:
I don't mean system path, I mean one of the CMake config settings.

If you use windows, as I see now, you can partially follow [this guide](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Windows) (it walks over building Cura entirely on windows, so just pick the engine bits out there).

ArthurKretzer, 2020-04-01:
Okay, I've followed the guide you showed me from zero with the following changes:
 
$ set PROTOBUF_MINGW_SRC=d:/dev/protobuf-3.0.0/
$ set PROTOBUF_MINGW_INSTALL=%PROTOBUF_MINGW_SRC%/install_dir/
$ set ARCUS_DIR=d:/dev/libArcus-master/install_dir/lib/cmake/Arcus

Finally, I've used this CMake config settings from CuraEngine\build:

% cmake -DCMAKE_INSTALL_PREFIX=../install_dir -DCMAKE_BUILD_TYPE=Release -DArcus_DIR=%ARCUS_DIR% -DPROTOBUF_SRC_ROOT_FOLDER=%PROTOBUF_MINGW_SRC% -DPROTOBUF_LIBRARY=%PROTOBUF_MINGW_INSTALL%/lib/libprotobuf.a -DProtobuf_INCLUDE_DIR=%PROTOBUF_MINGW_INSTALL%/include -DPROTOBUF_PROTOC_EXECUTABLE=%PROTOBUF_MINGW_INSTALL%/bin/protoc.exe -G "MinGW Makefiles" ..

But now my problem changed to :

"g++.exe: error: D:devprotobuf-3.0.0/install_dir//lib/libprotobuf.a: No such file or directory"

I don't understand why since there is a file libprotobuf.a in this directory that was created after I made protobuf instalation.

rburema, 2020-04-01:
Again, those variables are not environment variables, they're supposed to be set for CMake. (You can see that it strips out the path-separators in the error message.) If you put them as arguments to the cmake command-line (`-DPROTOBUF_MINGW_SRC=...`) it should work.

Might I also suggest to use cmake-gui? You won't need to type out those long commands.

ArthurKretzer, 2020-04-17:
That's a nice suggestion, but cmake-gui seens harder for me to understand and make it work.

Good news is: after a few tries, I was able to compile and use CuraEngine.

I was doing something wrong with folder naming and references... After I've followed strictly the tutorial, as naming protobuf folder as "protobuf-3.0.0-mingw", no error messages appeared. The only thing different I made in this successful compiling was to use D drive instead of C.

Thank you for the help. Take care!

closed 2020-04-17 by ArthurKretzer

rburema, 2020-04-22:
With fresh eyes I can also see that you where missing the path separator just after the drive-name in the last error message you showed.

(It's moot anyway since it builds now. Good job.)

commit 246ed65c (2018-04-18): Also apply maximum resolution to layer outlines

Because after combining parts we might have high-resolution bits again, causing trouble on the printers and also with our polygon offsets.

Fixes #1229.