Ultimaker/CuraEngine #1230: Commandline CuraEngine slice misalignment trouble

narutorne, 2020-04-02:
Thank you for seeing.

This is my first issue submission.
I am a beginner, and my English isn't so good so feel free to ask me if there is anything unclear.

I am trying to slice with CuraEngine on the command line.

Slicing is completed, but if I check the gcode that is output,
I recognize a misalignment. (image)

Is the jsonfile setting wrong? Or have I failed to install CuraEngine?

I would be grateful if you know how to solve some similar cases.

* Jsonfile was created by slicing with Cura GUI and collecting parameters from cura.log.
In addition, I tried using the default fdmprinter.def.json and fdmextruder.def.json and found that gcode was also misaligned.
(The result of the misaligned position was different.)

The command, jsonf
[image]
[commandline-result.txt](https://github.com/Ultimaker/CuraEngine/files/4419665/commandline-result.txt)
[json-3Dmodels-gcode.zip](https://github.com/Ultimaker/CuraEngine/files/4419675/json-3Dmodels-gcode.zip)

ile and stl file when executed are shown below.

----------------------------------------------------------------

command

~/curabuild/CuraEngine/build $ CuraEngine slice -v -j ../jsonfile
s/test_ender3pro.def -o ../gcode/"output_test.gcode" -l ../3Dmodels/heart-twist-
med.stl

Ghostkeeper, 2020-04-06:
There is nothing wrong in particular with the definition file, most likely. There is something wrong with the model.

Cura's front-end normally places the model in the middle of the build plate. CuraEngine doesn't do that by default. I think you'll want to add `-s center_object=true` to your command (just after the `-j jsonfile` command).

closed 2020-04-06 by Ghostkeeper

narutorne, 2020-04-07:
Thank you for your helpful reply.
I tried your advice and my problem was solved.

I don't know how to say thank you.
This will move me forward.
Thank you again for your kindness.

[image]